TheMicDiet/chihiros-led-control #2: Feature: Home Assistant Integration

TheMicDiet, 2023-08-23:
https://developers.home-assistant.io/docs/creating_component_index

titilambert, 2024-04-24:
Hello !
Thanks for your work !
I have 2 chihiros eggs and your tool is working ! (even if it has only 2 ilghts (red and green)
I can probably help on the home assistant integration.
I have few questions for you
* how do you want to do it ?
  * as official integration, in the home assistant core source code ?
  * as a third part integration, usable with hacs ?
    * In this case, where do you want to put the code of this integration ?
      * in this repo ?
      * in another repo ?
      
Thanks again !

TheMicDiet, 2024-04-27:
Hi, I think we can add it in this repo as a third party integration with hacs

kingunlim, 2024-04-28:
i cant wait for the addon in home assistant

titilambert, 2024-05-01:
Hello ! 
I have an alpha first version of a custom integration which is working with my 2 tiny terrarium eggs !
[image]
[image]

@TheMicDiet Do you want to create a new repo ? Then I could push my code there to let people testing it. Once you did it, let me know, I will do a PR to add Chihiros icon/logo to the home assistant brands repo.  

For now, I have some doubts how the code will act with other Chihiros lights...

Also, do you know if the Chihiros lights are only RGB (and RB for tiny terrarium eggs) ? or there are some models RGBW/RGBWW ?

Thanks !

TheMicDiet, 2024-05-01:
Hi @titilambert, afaik, there are lights that are RGB, RGBW as well as single color (white) only.

titilambert, 2024-05-02:
@TheMicDiet @kingunlim I need your opinion:
I have 2 tiny terrarium eggs which are exposing ONLY red and green. So this is only `RG` of the `RGB`.
Should I create an RGB light entity in HA or two BRIGHTNESS lights (one for `red`, the other for `green`) ?
And I have the same question for any other Chihiros lights. I mean, for this kind of light, we do not care about the colour of the light, but we want to control each colour individual light (one `red`, one `green`, one `blue`, maybe one or two `white`)?
Thanks for your answers.

TheMicDiet, 2024-05-02:
Hi @titilambert, I have no RGB light so it doesn't really matter to me. However, if you ask me, it would make sense to have separate controls for each color, as long as it doesn't get too complicated. Feel free to implement as you prefer it.

kingunlim, 2024-05-02:
> @TheMicDiet @kingunlim I need your opinion:
> 
> I have 2 tiny terrarium eggs which are exposing ONLY `red` and `green`. So this is only `RG` of the `RGB`.
> 
> Should I create an RGB light entity in HA or two BRIGHTNESS lights (one for `red`, the other for `green`) ?
> 
> And I have the same question for any other Chihiros lights. I mean, for this kind of light, we do not care about the colour of the light, but we want to control each colour individually (one `red`, one `green`, one `blue`, maybe one or two `white`)?
> 
> Thanks for your answers.

so i have a chihiros wrgb2 so individual controlls would be very nice.
but im not entirely sure what can be possible?
in the app there are scenes and so on, but when i can activate the light and the last set light in the app is loaded that would be great enough, i thing a gradual shift is hard to make? so i mean when home assistant turn on the light maybe the light slowly turns to a specific percentage. (but as i write i think a automation can do this also?) thank you guys for your work!

titilambert, 2024-05-03:
Ok ! So I will do one monochrome light entity per colour. It will not be more difficult !

kingunlim, 2024-05-04:
@titilambert 
awesome, im curious, can i already add the addon in home assistant via hacks?

titilambert, 2024-05-04:
Not yet, we have to make te repo hacs compatible. I never did that since I'm using hacs...

kingunlim, 2024-05-04:
alright i asume you mean you dont use hacs? 
so then i will wait for an option to install it via homeassistant via an install option.
can you please write again if there is a option to install? best regards and thx again

titilambert, 2024-05-04:
Sorry, I confirm I do not use hacs.
To help you I need to know more about your HA installation. Can you give me more details ?
Or if you can wait few more days I should be able to make it hacs compatible.

kingunlim, 2024-05-05:
of corse no problem i can wait till you are ready 👍

titilambert, 2024-05-05:
Hello ! my fork/branch is ready for a first HACS test: https://github.com/titilambert/chihiros-led-control/tree/ha_integration
I do not know how you can put a git repo URL and branch in HACS, but here is the info I can give:
* repo url: `https://github.com/titilambert/chihiros-led-control.git`
* branch (or release): `ha_integration`

Let me know if you have any questions/issues/comments !

Note: this will change when it will be merged to this repo

kingunlim, 2024-05-05:
i dont know but i think something is wrong [image]

kingunlim, 2024-05-05:
maybe this is the problem? https://github.com/hacs/integration/issues/2754

kingunlim, 2024-05-05:
[image]

titilambert, 2024-05-05:
@kingunlim did you had a chance to select the `ha_integratoin` branch ?
If not, this error is completely normal.
Let me know, and I will do something to get my main branch working

kingunlim, 2024-05-05:
im sry i dont know how i coild do that?
normaly i go in hacs and select user repository and the i insert the url an choose the category like this:
[image]

titilambert, 2024-05-05:
@kingunlim can you retry ? I just merged on my main branch. Just try with:
* repo url: https://github.com/titilambert/chihiros-led-control.git
* branch: main

kingunlim, 2024-05-06:
yes, now its working ;)
[image]

so, now i must learn to make the right actions to turn the lights on and off, 
when im in the app i can akticate the light and then i goes for example 70 white 70 red and 70 blue, now in your integration i must activate all channels manually?
or do you plan to update the integration for better communication with for example the wrgb2?
best regards!

titilambert, 2024-05-06:
Hello! First thanks for testing it ! I'm happy it is now working for you.
In fact, the app keeps some “profiles” about each color, like a shortcut.
I don't think I can do that in the integration since it will be different for each user.
So I would recommend creating a group of light, see https://www.home-assistant.io/integrations/group/#binary-sensor-light-and-switch-groups.
With that, I guess you will have the behavior you want to turn on/off the light quickly.

PR #16 (2024-05-09): Refactoring to add HA integration code
This PR fixes #2 and #14 

Hello ! this PRs adds the HA integration !
So please feel free to test it ! and also don't forget to test also the `chihirosctl` CLI !
Thanks !
closed 2024-05-09 by TheMicDiet