TheMicDiet/chihiros-led-control #3: Thanks!

stigi, 2023-11-26:
Hi @TheMicDiet,

Just wanted to say thanks for the repo, especially the extensive Readme.

I've just got my first fish tank and was planning to automate the lighting via Alexa and then only discovered that my AII401 speaks bluetooth ❤️ ([not much advertised when I bought it](https://dennerle.com/produkte/aquarium/nano-tank/nano-tank-plant-pro/)).

So now that I know It speaks BLE I came up with the idea of building an Alexa skill. I think there's a nice prebuild voice preset that I could connect to the dimmer: [PowerLevelController](https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-powerlevelcontroller.html).

I'll keep you posted on my progress, but didn't want to wait with giving you a shout out. :) 

🙌

TheMicDiet, 2023-11-27:
Hi @stigi,

I'm glad to hear that my project is helpful for you! When I bought the LED, I couldn't find any information about the BLE protocol, so I thought it would be a good idea to share my findings with future tinkerers. 

Good to know that Dennerle kits also come with Chihiros LEDs, I bought mine separately. 

Good luck with your Alexa integration. I wanted to build a Home Assistant addon, but didn't have the time yet ;)

closed 2023-11-27 by TheMicDiet

kingunlim, 2024-03-04:
are you still planimg to do a home assistant addon?
i also have the wrgb2 and that would be so awesome to have it in my house automations.
best regards

stigi, 2024-03-05:
I got demotivated because of the state of Alexa dev docs. I was hoping to establish a bluetooth connection between my Alexa device and the light, but to no success.

I ended up just getting a Chihiros Wifi hub for dirt cheap (https://www.aquasabi.de/Chihiros-Wifi-Hub) and can control my lights now from outside of bluetooth reach. Automation still only works via the Chihiros app.