TheMicDiet/chihiros-led-control #8: How to connect to the module from PC?

cybericius, 2024-02-25:
Hi,
I have a Chihiros WRGBII module. When I want to add it through Bluetooth, discovery doesn't see it on PC (neither on Linux or Windows). Tried it while the phone was off, where connection works through the official app. 
Should I reset it in some ways? 
Thanks.

cybericius, 2024-02-25:
(venv) akos@xray:~/python/chihiros-led-control$ python3 ./chihirosctl.py list-devices
╭─────────────────────────────── Traceback (most recent call last) ────────────────────────────────╮
│ /home/akos/python/chihiros-led-control/./chihirosctl.py:25 in list_devices                       │
│                                                                                                  │
│    22 @app.command()                                                                             │
│    23 def list_devices(timeout: Annotated[int, typer.Option()] = 5):                             │
│    24 │   table = Table("Name", "Address")                                                       │
│ ❱  25 │   discovered_devices = asyncio.run(BleakScanner.discover(timeout))                       │
│    26 │   for device in discovered_devices:                                                      │
│    27 │   │   table.add_row(device.name, device.address)                                         │
│    28 │   print("Discovered the following devices:")                                             │
│                                                                                                  │
│ ╭─────────────────────── locals ────────────────────────╮                                        │
│ │   table = <rich.table.Table object at 0x7fa7254da9b0> │                                        │
│ │ timeout = 5                                           │                                        │
│ ╰───────────────────────────────────────────────────────╯                                        │
│                                                                                                  │
│ /usr/lib/python3.10/asyncio/runners.py:44 in run                                                 │
│                                                                                                  │
│   41 │   │   events.set_event_loop(loop)                                                         │
│   42 │   │   if debug is not None:                                                               │
│   43 │   │   │   loop.set_debug(debug)                                                           │
│ ❱ 44 │   │   return loop.run_until_complete(main)                                                │
│   45 │   finally:                                                                                │
│   46 │   │   try:                                                                                │
│   47 │   │   │   _cancel_all_tasks(loop)                                                         │
│                                                                                                  │
│ ╭──────────────────────────────── locals ────────────────────────────────╮                       │
│ │ debug = None                                                           │                       │
│ │  loop = <_UnixSelectorEventLoop running=False closed=True debug=False> │                       │
│ │  main = <coroutine object BleakScanner.discover at 0x7fa7255ffbc0>     │                       │
│ ╰────────────────────────────────────────────────────────────────────────╯                       │
│                                                                                                  │
│ /usr/lib/python3.10/asyncio/base_events.py:649 in run_until_complete                             │
│                                                                                                  │
│    646 │   │   if not future.done():                                                             │
│    647 │   │   │   raise RuntimeError('Event loop stopped before Future completed.')             │
│    648 │   │                                                                                     │
│ ❱  649 │   │   return future.result()                                                            │
│    650 │                                                                                         │
│    651 │   def stop(self):                                                                       │
│    652 │   │   """Stop running the event loop.                                                   │
│                                                                                                  │
│ ╭─────────────────────────────────────────── locals ───────────────────────────────────────────╮ │
│ │   future = <Task finished name='Task-1' coro=<BleakScanner.discover() done, defined at       │ │
│ │            /home/akos/python/chihiros-led-control/venv/lib/python3.10/site-packages/bleak/_… │ │
│ │            exception=BleakDBusError('org.bluez.Error.InProgress', 'Operation already in      │ │
│ │            progress')>                                                                       │ │
│ │ new_task = True                                                                              │ │
│ │     self = <_UnixSelectorEventLoop running=False closed=True debug=False>                    │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                                  │
│ /home/akos/python/chihiros-led-control/venv/lib/python3.10/site-packages/bleak/__init__.py:242   │
│ in discover                                                                                      │
[... cut ...]
│ │ reply = <Message ERROR serial=358 reply_serial=8 sender=:1.9 destination=:1.182212 path=None │ │
│ │         interface=None member=None error_name=org.bluez.Error.InProgress signature=s         │ │
│ │         body=['Operation already in progress']>                                              │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────╯ │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress

TheMicDiet, 2024-02-25:
Hi, maybe your Bluetooth adapter is not compatible. However, you can try to delete the Chihiros device in the app and restart your PC to resolve this operation already in progress error.

closed 2024-04-27 by TheMicDiet