TheMicDiet/chihiros-led-control #13: Autodetect device model

titilambert, 2024-04-27:
Hello, I'm wondering how de app is able to know the device model.
I have only 2 tiny terrarium eggs, and here are they names:
* `DYDDF3CB70832898`
* `DYDDF733C80AF1CC`
I know that the 4 characters are the same : `DYDD` and the rest is the device mac address.
So I'm wondering if `DYDD` means `tiny terrarium egg` and if it's the same for other models, like the first 4 characters define the model.

If it's the case, we could hard-code a map to make the link between the first 4 characters and the model name.

Thanks !

TheMicDiet, 2024-04-27:
Yes, I think the first characters encode the device model and the last 12 characters are for the MAC address. In my case I have Chihiros LED A2 with the name `DYNA2 XXXX XXXX XXXX`

boominger, 2024-04-27:
I have a WRGB II and it has the prefix `DYNWRGB` followed by its mac address. I am currently waiting for my Universal WRGB and will lookup its name and test if it is working when it arrives.

closed 2024-05-09 by TheMicDiet