TheMicDiet/chihiros-led-control #21: Support WRGB 2 Pro

olapettersson, 2024-05-12:
I have a WRGB 2 Pro, which is not discovered. My A2-lights are though. The model is DYWPRO, (without the 'N' after 'DY'). I think this light is the same as the DYNWRGB, only more powerful. Would it be supported by just adding a file for it under the directory 'devices'?
Kind regards

TheMicDiet, 2024-05-12:
Yes, you can create a new config for your device in the device directory and add the new model name prefix to the manifest `custom_components/chihiros/manifest.json`.

olapettersson, 2024-05-12:
I got it to work, sort of, but when changing the brightness of each og the channels, it almost immediately turns off again. But the colour is correct. If you have any ideas, I'd be happy to try them out.

TheMicDiet, 2024-05-13:
@olapettersson did you solve the issue? As I understand it, you set the brightness of a single color and the LED turns on for a short time and turns off without any further interference. Is that right?

olapettersson, 2024-05-13:
That's how it behaved after I manually added a config for WRGB2Pro modified the manifest file, but I have not tried since you added support in the repo. Will try tomorrow.

olapettersson, 2024-05-14:
Ok, I tested it, and it behaves correctly. Must have been the fact that last time I tested, my automation in the Chihiros-app was still active, and it was outside of the hours for which the light was supposed to be on and hence it tuned off immediately.

However, when adjusting the white-channel, I only get red. Blue and green works correctly. Do these lights actually have dedicated white LEDs, or do the use RGB to render white light? If they only have RGB-LEDs, white should probably set them all to an equal level when adjusted. If the DO have white LEDs- the controller for white is adjusting the wrong channel.

I also have a couple of A2, and the only have white light, but are presented with controllers for RGBW. Both red and white control the white light, whereas blue and green do nothing. I don't know if this provides a clue as to how the channels work. In any case, If possible, the controllers for RGB could likely be removed from A2-lights.

olapettersson, 2024-05-15:
Update on the white channel of the WRGB II Pro. Someone actually measured the output spectrum and could confirm that there is a white LED in addition to RGB, which wasn't the case with the non-Pro version, where the white channel was simulated with the RGB-diodes. Link here: https://www.ukaps.org/forum/threads/chihiros-wrgb2-pro-has-a-separate-white-led-channel-but.68878/

Now, how can we find out how to control this channel? As of now, the white channel seems to control red. Please let me know if you would like me to test with my light. I know a little Python so I can probably hack around a bit if you give me some pointers at to where.
Thanks

TheMicDiet, 2024-05-16:
Unfortunately, the old app which I used for getting the info about the commands only had commands for RGB and single-color LEDs. I think the best way to find out how to control the white channel would be to sniff the Bluetooth packets that are sent to the LED when you manually set the white channel via the App and then we could try to decode them.

olapettersson, 2024-05-16:
I was thinking that only the colour parameter needed changing? Red: 0, green: 1, blue: 2 so perhaps white: 4? But I assume it is more complicated than that.
Do I need root access to fetch the HCI-log?

TheMicDiet, 2024-05-16:
No, you don't need root for snooping. You could try out different channels by changing the number of white in `custom_components/chihiros/chihiros_led_control/device/wrgb2_pro.py`. What happens if you increase blue brightness in the current implementation?

olapettersson, 2024-05-16:
Red,  green and blue works correctly, despite being `1`, `2` and `3` respectively in `wrgb2_pro.py`. I thought they were `0`, `1` and `2`?
I tried changing `white` from `0` to `4`, just to test, restart Home Assistant, but the white control till controlled the red channel. Will try other values, if that could make sense?

I will also try snooping. Where do I find the file? Only via ADB?

TheMicDiet, 2024-05-16:
For RGB they are 0,1 and 2. Are you sure that the changes were picked up correctly? 

I think I did download the snoop logs with ADB, yes.

olapettersson, 2024-05-16:
I thought these values should correspond to RGB bein 0,1,2, but maybe I have it backwards?
[image]

TheMicDiet, 2024-05-16:
I am just wondering why 0, 1 and 4 would all change the red channel.

olapettersson, 2024-05-16:
Hmm, I will see if setting white to 2 or 3 will make it change the green and blue channels. BTW, also tried "white: 5" which still controlled red.

But just to make sure. I make a change and then restart Home Assistant, and the change should be applied? Just so that possible "carbon based errors" are eliminated. 🤪

olapettersson, 2024-05-18:
No matter what I change the numbers in wrgb2_pro.py to, the colour it changes is the "name" of the colour (except for white, which changes red). I expected the blue channel to adjust red if I set blue: 0, but it still changed blue. It is as if these numbers don't mean anything.

TheMicDiet, 2024-05-18:
Hm, seems weird. Maybe you can try to use the `set-brightness` command directly from the CLI and change the color: https://github.com/TheMicDiet/chihiros-led-control/blob/main/custom_components/chihiros/chihiros_led_control/chihirosctl.py#L83C42-L83C50)

olapettersson, 2024-05-18:
Ah, I noticed something! My WRGB 2 Pro is discovered as "fallback", not actually WRGB 2 Pro. In ’fallback.py’, both red and white are ’0’, which probably explains what I experience. Will try changing white to other values tomorrow.

But why is the light not discovered correctly? In fact, my A2s are also discovered as fallback, which can explain why they all have WRGB controls.

My devices' model codes are:
A2:
DYNA2N + mac-address, works correctly using the white channel.

WRGB2PRO:
DYWPRO90 + mac-address
I didn't think 90 mattered, as it is the length of the light, but maybe it does?

olapettersson, 2024-05-19:
Ok, I figured It out! 😃

Since the light is erroneously discovered as "fallback" (see previous comment), it is in fallback.py I had to experiment. And the answer is easy! 😅

Red: 0
Green: 1
Blue: 2
White: 3

TheMicDiet, 2024-05-19:
Nice! I will release a new version with a fix. Basically, the last 12 chars are removed from the device name and the rest are used to match the device.

TheMicDiet, 2024-05-19:
@olapettersson can you try out if version 0.2.1 works for you?

TheMicDiet, 2024-05-19:
I am also wondering if the white channel on the base WRGB models is really 0. @kingunlim does the white channel of your WRGB II work correctly with the integration?

olapettersson, 2024-05-19:
> @olapettersson can you try out if version 0.2.1 works for you?

Will do! Thanks for fixing it so quickly!

closed 2024-05-19 by TheMicDiet