TheMicDiet/chihiros-led-control #24: Feature Request: Support for Wifi Hub

txyaloo, 2024-05-16:
Any chance you might be able to support the wifihub?  It's ESP07 based,  I think that would be the easiest way to control multiple lights.

I've tried to sniff the wifi packets and dump the firmware but haven't had much luck.   I was able to get a scrambled uart signal from the ESP chip itself, but I couldn't get anything from what looked like UART pins on the chihiros board. 

Board pics:

[image]

[image]

[image]

TheMicDiet, 2024-05-16:
Hi @txyaloo, I only have one LED A2 and therefore no need for the hub, so I won't be able to reverse engineer it.

olapettersson, 2024-05-16:
Hijacking this a bit 😛 
@txyaloo I too have the WiFi hub but it does not work reliably. Lights do not turn on, or they suddenly turn off. I have one, placed reasonably in the middle of My A2s and WRGB II Pro. I also can't seem to reset it. I says in the manual that to reset, you hold down on the Chihiros logo, but that doesn't seem to work. Would you mind posting a picture of the other side of the PCB, so I can see if there is a switch button there?

In any case, this library/integration is working remarkably well so far, despite being in a very early stage. Only issue now is to figure out how to control the white channel on the WRGB II Pro. So I will not be using my WiFi hub anymore.
Thanks!

txyaloo, 2024-05-16:
> Hi @txyaloo, I only have one LED A2 and therefore no need for the hub, so I won't be able to reverse engineer it.

Completely makes sense.  Could you add support for the WRGB2 that starts with DYNW natively?  I was able to add to the manifest.json and clone your existing wrgbii.py driver to get it to work.

I have two of these WRBII DYNW lights.  One is DYNW3 and the other DYNW6.  Both WRGB2

TheMicDiet, 2024-05-16:
Yes, we can do that, see #25. Are DYNW3 and DYNW6 the full prefixes, i.e. all characters but the last 12?

txyaloo, 2024-05-16:
Full value is DYNW30 and DYNW60.

txyaloo, 2024-05-16:
> Are these 30 and 60 cm LEDs?

I assume the 30 and the 60 is the length?  That would make sense.  The 30
is ~12" and the 60 is ~24"

Looks like the 60 is actually a WRGB2 Pro, but it isn't listed that way in
the app.  This is the specific light I have.

https://www.amazon.com/gp/product/B09YQXLX2T/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

TheMicDiet, 2024-05-19:
Hi @txyaloo, can you test if version 0.2.1 of the integration works for you. It would also be interesting to know if the white channel is functioning correctly.