TheMicDiet/chihiros-led-control #31: Device password

milsky99, 2024-06-13:
First off thanks for the work in getting this integration done

I have failed at installing the integration, it is asking for a device password, I have left this bland and it returns an error, other passwords don't seem to work. May just be something Im overlooking but assistance would be appreciated

Great work

D

milsky99, 2024-06-13:
putting the logs in here

Logger: aiohttp.server
Source: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
First occurred: 12:00:15 PM (6 occurrences)
Last logged: 5:12:29 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 88, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp_session/_init_.py", line 199, in factory
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 295, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 71, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 368, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 414, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 518, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/chihiros/config_flow.py", line 161, in async_step_user
    assert self._discovered_device is not None
AssertionError

TheMicDiet, 2024-06-13:
Looks like there is no compatible device discovered. Are you sure that your device is in range? What device are you trying to connect?

milsky99, 2024-06-13:
Thanks for this, im trying to connect a slim v2 BUT this is not even installing the integration and happens immediately, not even a split second of a delay

TheMicDiet, 2024-06-13:
Did you set password for your device in the chihiros app?

milsky99, 2024-06-13:
I don't recall doing so, I searched for any password BUT did not find any

I have to be honest this did not look like there was any querying of any app or device it was literally instantaneous.... It appeared to me as if the integration was failing rather than the process of querying any device, I then powered down the lights and was 100% exactly the same. Could this not be an issue with the installation of the integration ?

TheMicDiet, 2024-06-13:
Well, from the error above it seems like no compatible device is discovered. The discovery happens in the background, so thats why the manual setup step fails immediately. Could you provide the name of your device, maybe it is currently not supported.

TheMicDiet, 2024-06-13:
What you could also try is to disconnect the LED from your phone, maybe this could help.

milsky99, 2024-06-13:
Im going to do some more work here and come back to you, thanks for the input, greatly appreciated

TheMicDiet, 2024-06-14:
I released a new version which has an improved manual setup step that let's you select the discovered devices. maybe you can try it out and see if your device shows up.

milsky99, 2024-06-17:
Thanks a ton for this !!

milsky99, 2024-06-19:
Ok so retried the integration after installing the last three recent updates and BOOM there it was. Picked up my 3 lights and entities and we’re off. Once again thanks for the great work

closed 2024-06-20 by TheMicDiet