TheMicDiet/chihiros-led-control #35: Disconnections

milsky99, 2024-07-22:
Im running a bluetooth proxy on ESP32, all installed correctly, the Integration is also running and discovers my 3 lights however the integration seems to disconnect from the light for no reason. 

At the moment 2 of the 3 lights are working, all 3 are showing as connected but the one simply doesn't respond to any HA instructions.

Any clue how I can troubleshoot this, could it be the proxy ?? Not sure where to start looking

Thanks again for all the work !

TheMicDiet, 2024-08-03:
Hi @milsky99, unfortunately I have no experience with running a bluetooth proxy.

taky-taky, 2024-12-25:
Hi @milsky99 

I've done some research that might help you.
First, ESP (Bluetooth proxy) has a limit on how many active connections can be made in parallel.
[](https://esphome.io/components/bluetooth_proxy.html)

```
The Bluetooth proxy of ESPHome provides Home Assistant with a maximum number of 3 simultaneous active connections. Devices which maintain a continuous active connection will consume one of these constantly, whilst devices which do periodic disconnections and reconnections will permit using more than 3 of them (on a statistical basis). Passively broadcasted sensor data (that is advertised by certain devices without active connections) is received separately from these, and is not limited to a specific number.
```

If your ESP uses ardiuno framework you are in a worse situation (as I am.)

```
The [ESP32 Platform](https://esphome.io/components/esp32) component should be configured to use the esp-idf framework, as the arduino framework uses significantly more memory and performs poorly with the Bluetooth proxy enabled.
```

I cant switch because the device needs the ardiuno framework for other reasons.

As I can see chihiros-led-control DO maintain a continuous connection (I am not able to controll anymore the leds from the official app, after issuing a command from chihiros-led-control)
My solution is to reload the integration for the device in ESPHOME which is responsible for ble_proxy.
This releases the connection.
Also dont use long runing scripts in home assitant to controll the led, like https://community.home-assistant.io/t/increase-or-decrease-light-brightness-over-time/600349
If the ble connection disconnects your sript will abort.
Just short, prompt commands.

closed 2025-09-27 by TheMicDiet