TheMicDiet/chihiros-led-control #38: Unable to add WRGB 2 10th edition

HannibalChau, 2024-08-30:
Hello,

I'm having an issue adding WRGB 2 10th edition.  Would you be able to add it?  Device id: DYNW12PD9EC8584B299

Thanks

senna1992, 2024-08-30:
I did connect with the standard wrgb ii successfully

 here is what I would check in your case :

--> is the bt module from chihiros placed next to interferencing stuff?

--> do you use the internal antenna of the esp or external (stick on type works better for me compared to screw on with adapter cable)

--> what firmware is your light on?

--> do you have more then 3 ble light / devices? If so get another bt proxy installed :)

Hope that helps

TheMicDiet, 2024-08-30:
Hi @HannibalChau, is this the 120 cm version?

HannibalChau, 2024-08-30:
Yes, firmware 22.

HannibalChau, 2024-08-31:
That worked, thank you for the quick response. Now I just need to figure out an automation to do.  :)

senna1992, 2024-08-31:
Claude and chatgpt.com are very helpful for that

closed 2024-08-31 by TheMicDiet