TheMicDiet/chihiros-led-control #39: Support for WRGB 2 SLIM

joshua-dessol, 2024-08-31:
Hey there!

Could you add the WRGB 2 SLIM device ID to the source code please?

ID: DYSILNFFCC19826581

It's the 45cm version.

Really appreciate it! Have a nice day.

TheMicDiet, 2024-08-31:
Sure, done. Check out 0.6.0.

jonathanalvin, 2024-09-01:
My Chihiros unable to connect, do you need to specify device ID somewhere? I have the same device but different Device ID.

senna1992, 2024-09-01:
> My Chihiros unable to connect, do you need to specify device ID somewhere? I have the same device but different Device ID.

can you see DYSILNFFCC19826581 when pressing "add new entry"?
if not, did you remove the light from the chihiros app?
if yes, did you try to place the BT-transmitter closer to the light's bt-controller?

jonathanalvin, 2024-09-01:
I saw my own ID on the Chihiros App. Do I need to unpair it from my app first? My light is built in BT Controller.

senna1992, 2024-09-01:
Yes unpairing from chihiros app is required to migrate to this integration

joshua-dessol, 2024-09-11:
Sorry for the late response, I was testing it the last few days and its working flawlessly, thanks a lot!

closed 2024-09-11 by joshua-dessol