TheMicDiet/chihiros-led-control #43: Support for B series

IanVan5, 2024-09-04:
Hello!

The device ID of my B90 is DYCOMCDD1CD025385

I believe the ID conflicts with the model code used for the backlight, which is "DYCOM". I was able to add the device as a result and can turn the white LEDs on and off.

TheMicDiet, 2024-09-05:
Hi @IanVan5, you are right. It seems like DYCOM is the model code of the "Commander 1" which can be used to control a wide range of devices (A, B, C...). I will release a new version to address this issue. Unfortunately, it won't be possible to detect which device is controlled by the commander, so it will be generic.

IanVan5, 2024-09-10:
I’ve tested the latest version, the red toggle controls the white LEDs and none of the others do anything. I will try playing with the command line tool when I get some time.

senna1992, 2024-09-10:
Please check the channel association in the device descriptions. They are saved locally after installing (custom components/chihiros/device)

White should be channel 0 instead of 3.
Red green and blue can be deleted from your descriptor since the light only has brightness controls (if it is commander 1 based).

After changing the file (iE using file editor)
try reloading the light in the integration or restart HA.

Please provide full Mac address + corrected device descriptor for your light so that the author of this integration can identify a good identifier and link it to identical lights.

Thanks :)

0 should be

senna1992, 2024-09-10:
@TheMicDiet do you think it is possible to add a manual profile selection option?

iE ccw, brightness only, RGB, wrgb

with the corresponding channel association predefined (for commander 1 channel 0 is brightness / white)

IanVan5, 2024-09-11:
I see that the RGB lights turn on with the white LEDs now, I was under the incorrect impression that they could be turned on separately.
MAC address: CD:D1:CD:02:53:85
@senna1992 what is the "corrected device descriptor" and where can I find it?

TheMicDiet, 2024-09-11:
> @TheMicDiet do you think it is possible to add a manual profile selection option in the configuration process ?
> 
> iE ccw, brightness only, RGB, wrgb
> 
> with the corresponding channel association predefined (for commander 1 channel 0 is brightness / white) 

I think it should be possible, I have thought about doing the same thing for the fallback device before.