TheMicDiet/chihiros-led-control #44: WRGB II Slim - Unable to find via bluetooth

jonathanalvin, 2024-09-11:
My Device ID DYSILNF5EOACC1EC37 is not appearing in Chihiros integration while searching.

I have deleted the light from MyChihiros app but still could not find it in HASS.

Is there anything specific I should do after deleting the device from the app and before pairing it with HASS?

scweiser, 2024-09-12:
I'm having the same issue

Backkup, 2024-09-15:
Hi, same issue here, i'm thinking it may be than my HA serveur can't use bleutooth ? (not sure about that)
Also i can't install the python environment (I'm not the brightest whit those)

senna1992, 2024-09-15:
For installing python environment check out esphome tutorials for your operating system (mac.linux,winows). its also python based and well documented. for short questions, chatgpt is helpful as well 🙂

TO be able to use BT from your Server, it has to have bluetooth - and therefore will show it as new detected device. (did you disable/ignore it long time ago maybe?)

can you see other bt devices from within HA?
did you enable BT-Integration and select the correct device?

also i BT-Range is not great on my CM4 based HA yellow. i added additional esp32 based BT-Proxys, work like a charm after learning to set them up and configure.

I can see my lights from my phone with their BYLE or whatever the name is. so after having deleted it form chihiros app you should check that first!

jonathanalvin, 2024-09-16:
I can find some devices were listed down from the integration page.

My bluetooth is around 5 meters away and when I am using the app, I can still get connected although I stood further than 5 meters, hence I believe range should not of an issue in this case.

Just wondering, when you discover the device from HA that you can exactly find the device like DYSxxxxxx?