TheMicDiet/chihiros-led-control #49: Getting state?

CoreyJ87, 2024-11-20:
This is less of an issue and more of a question for micdiet. I've forked this repo. As I want to update the hacs plugin to do alot more. I've made some progress on adding more entities to control more stuff on the device but I'm not seeing any state stuff in the underlying library. Which makes it kind of hard to add the things I want to. As you have no clue whether the command actually did anything. It just fires off and assumes. 

Has anyone been able to get data back from the light about what it's doing? I've been playing with the characteristics and notifiers on my WRGB 2 pros but I can't get anything to come back. Commands all work but I never get states back. Curious to see if anyone's played around with this before or knows any info about how the chihiros app knows it's state? Thanks

TheMicDiet, 2024-11-20:
see https://github.com/TheMicDiet/chihiros-led-control/issues/36

CoreyJ87, 2024-11-22:
> see https://github.com/TheMicDiet/chihiros-led-control/issues/36

Ah okay. Dang. Guess it just has to fire off and assume. Thanks

closed 2024-11-22 by CoreyJ87