TheMicDiet/chihiros-led-control #51: Command line issue?

toofewacres, 2024-11-21:
### Discussed in https://github.com/TheMicDiet/chihiros-led-control/discussions/50

<div type='discussions-op-text'>

<sup>Originally posted by **toofewacres** November 20, 2024</sup>
So far I have installed the Chihiros integration. I was able to add my 2 lights and they show up as lights. 
I really want to be able to control them thru HA as far as ramp up times, on off times ,etc.  I see in the readme file the info about using the CLI. I tried following the info but had issues right of the bat when I tried to do the "pip install -e ." command. Here's the screen info:

 venv) ➜  chihiros source venv/bin/activate
(venv) ➜  chihiros pip install -e .        
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Obtaining file:///homeassistant/custom_components/chihiros
ERROR: file:///homeassistant/custom_components/chihiros does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

What directory should I be trying to run this in? I've tried at several directories, but get the same error.  I've tried searching for the "setup.py" file, but don't find it anywhere. I'm green at the CLI side (actually linux as well) and can follow directions, but when things don't work, I'm kind of lost. Any help would be great.

I'm running 
  OS Version:               Home Assistant OS 13.2
  Home Assistant Core:      2024.11.2
with ➜  ~ python             
Python 3.12.3 (main, Aug 23 2024, 06:10:48) [GCC 13.2.1 20240309] on linux
</div>

TheMicDiet, 2024-11-21:
You should be able to run "pip install" in the root dir of this repository. To turn the lights on and off automatically you can use Home Assistant Automations.

toofewacres, 2024-11-21:
1st of all, thank you for pulling this all together.  Chihiro's' WIFI 
hub is the pits. I had 2 of them over about a year and they just quit 
working. Plus their app is not the easiest to deal with either.

I have just about every other aspect of of my tank monitoried or 
controlled thru HA and my lights were the missing part..

  Adding a bluetooth proxy with an ESP32 and ESPhome is very simple and 
works great. Only about $10 costs too.

The functions for the lights work fine. On, Off, brightness adjustment. 
etc.

One thing is that my lights show up as using the "fallback" .

I have a WRGB Pro 120 it's id is DYWPR120   which is different than what 
is in your files. The fall back configuration works.

I laso have a z mini which shows up as DYZSD (which I thing is what you 
are referring to as the terrarium light)  It only has 2 channels to 
control.

I also tried  again with the command line. There seems to be something 
not working when pip is trying to install. Must be that HA doesn't have 
all of the proper  python files in their distribution.  I'll do some 
digging and see if I can sort it out..

The only reason I was thinking I want to be able to use the CLI is for 
setting scheules and ramp up times. I know I can go the route thru 
writing some automations scripts and accomplish the same thing so if I 
can't figure it out thru CLI, I'll go that route.

Thank you again.

Links:
------
[1] 
https://github.com/TheMicDiet/chihiros-led-control/issues/51#issuecomment-2490150097
[2] 
https://github.com/

TheMicDiet, 2024-11-26:
Hi @toofewacres, thank you for the info I fixed the model code for the WRGB II PRO 120, so it should be detected correctly. I also added the possibilty to connect the Z Tiny LED using fallback. If you want full support for Z Tiny I need to know what color in the fallback controls which channel of your LED.

toofewacres, 2024-11-29:
Thanks Michael. The update picks up the wrgb pro 120 correctly now. I'll get you the Z tiny info over the weekend.

closed 2025-05-24 by TheMicDiet