TheMicDiet/chihiros-led-control #57: Support for heater pro

phaeton82, 2025-02-11:
A bit out of scope for a led light. But Chihiros launched a heater. Which also can be controlled via bluetooth. 

Maybe an idea to support in the future?

https://www.chihirosaquaticstudio.com/products/chihiros-heater-pro-eu-version

TheMicDiet, 2025-03-11:
Hi, unfortunately I don't plan to support the heater as I don't own one myself and it would be harder to reverse engineer the protocol because this is a newer device I think.

TheMicDiet, 2026-09-20:
I added support for the heater based on decompiled APK. It would be nice, if you could test it and give some feedback.

phaeton82, 2026-10-25:
@TheMicDiet 

I have bought the heater Pro and installed it. I would like to be of an assistance to help the support of this device in this integration. Hereby my first impressions:

The Heater is being found automatically in Home Assistant
After adding the device in Home Assistant the Chihirios App cannot find the device anymore. The connection to the heater is being blocked by the integration. After disabling the device, the heater is usable in the app again.

For some reason the values of the device are not updated in Home Assistant when you alter them via the integration. When I change values via the integration I also do not see them changed in the chihirios app (after disabling the device in Home Assistant).

So for now the Heater Pro is not really supported in the integration. Can I help you with integrating this better? 

For now I have tried to capture the packets. Enclosed the CPA file I got from my phone. I did the following in the Chihiros app:

step 1: choose Auto (from manual to auto)
step 2: Back to manual
Step 3: Set temperature to 26 (from 25 degrees celcius)
Step 4: Set temperature to 24
Step 5: Set temperature back to 25 (the value I started with)
Step 6: Set power to 400w (from 200 watts)
Step 7: Set power to 600w
Step 8: set power to 800w
Step 9: set power to 1000w
Step 10: set power to 200w

[btsnoop_hci_202610251957.cfa.zip](https://github.com/user-attachments/files/33936428/btsnoop_hci_202610251957.cfa.zip)

TheMicDiet, 2026-10-31:
Hi @phaeton82, thank you for the feedback. Could you please test the latest version and check if the issues are resolved? Using the app and the integration in parallel won't really work since you can't read all settings from the device. The official app keeps these settings stored locally.