TheMicDiet/chihiros-led-control #58: Missing Blue Color Control in Home Assistant Integration for Tiny Egg

Tuatop, 2025-03-07:
I successfully set up the Chihiros Tiny Egg using the Home Assistant custom integration via HACS. However, I noticed that only two colors (Red and Green) are controllable (On/Off), while Blue is missing.

Is this expected behavior, or does the integration need improvement?

Thank you!

TheMicDiet, 2025-03-11:
Hi, it's on purpose because the first person to request support for the Chihiros tiny egg said there are only two channels. if there are actually 3 channels I can fix that. Maybe it's a different version of the egg.

Tuatop, 2025-03-11:
Hi, I appreciate your quick response! I checked again, and you’re right. My Tiny Egg also has only two controllable channels. I misunderstood at first, so I mistakenly posted the issue. I sincerely apologize for the confusion and appreciate your work on the integration.

Thanks again!

closed 2025-03-11 by Tuatop