TheMicDiet/chihiros-led-control #59: Support Brightness Control for Chihiros Tiny Egg

Tuatop, 2025-03-11:
Hi, First, thank you for your work on this integration! I successfully set up the Chihiros Tiny Egg using the Home Assistant custom integration via HACS. Currently, I can turn the lights (Green, Red) on and off, but I don’t see an option to control their brightness percentage.

Is brightness adjustment something that could be supported in a future update? I’d love to be able to fine-tune the light intensity.

Thanks again!

TheMicDiet, 2025-03-11:
Hi, this should work already.

Tuatop, 2025-03-11:
Hi, I tested it again, and the brightness control actually works perfectly! The reason I thought it wasn’t working at first is that the initial brightness percentage displayed on the Home Assistant dashboard was 1%, but in reality, the light was at 100%. That caused some confusion on my end.

I sincerely apologize for the mistake and appreciate your work on this integration. Thanks for your patience!

closed 2025-03-11 by Tuatop