TheMicDiet/chihiros-led-control #67: Support for Chihiros Dosing Pump System?

MankiniChykan, 2025-06-12:
Is it possible to add support for the Chihiros Doser Unit?

https://www.chihirosaquaticstudio.com/collections/chihiros-dosing-pump-system

TheMicDiet, 2025-06-15:
Hi @MankiniChykan, this could be possible, but someone would have to reverse engineer the protocol to control the pump. I don't own a pump myself, so I can't do that, but if someone has one and wants to help out, PRs are welcome!

MankiniChykan, 2025-06-16:
I have a 4 head doser. Willing to help, no idea where to start.

TheMicDiet, 2025-06-28:
Hi @MankiniChykan, thank you for your offer, but I am currently unable to look into the dosing pump. For the LED, I did sent commands via the official app and recorded the Bluetooth packages with the Android HCI Snoop Log. This log can be imported in Wireshark for further analysis. For the LED, it was also very helpful to look at the old Chihiros app to get some hints on how the commands work on the byte level. However, I don't think that the old app supported the dosing pump, so it will be harder to reverse engineer the command structure in this case. Maybe it's similar to the LED protocol.

Martin11180, 2025-09-06:
Hello
I have two dosing pumps and would like to help get the pump working with its code.
I don't know if it's long enough as an attachment.
I've changed the MAC ID.
greet 
Martin

[export.txt](https//github.com/user-attachments/files/22192129/export.txt)

Martin11180, 2025-09-18:
Hello
I've made some progress

Can someone test this code
in a Python environment?
It may be that some modules need to be installed via pip.

Sorry for this code, it's my first Python code, but it works for me.

It would be good if the developer's code had the option to pass the Command ID, mode, and Command Parameter parameters for testing.

filename test.py with this code

```
import asyncio
import argparse
import typer
import struct
from typing import Any
from bleak import BleakClient
from bleak.exc import BleakError
from datetime import datetime
from rich import print
from rich.table import Table
from typing_extensions import Annotated
from prettytable import PrettyTable
from prettytable import TableStyle

uart_service = '6E400001-B5A3-F393-E0A9-E50E24DCCA9E'
uart_rx      = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E'
uart_tx      = '6E400003-B5A3-F393-E0A9-E50E24DCCA9E'

def remove_bytes(buffer, start, end):
    fmt = '%ds %dx %ds' % (start, end-start, len(buffer)-end)  # 3 way split
    return b''.join(struct.unpack(fmt, buffer))

def next_message_id(current_msg_id: tuple[int, int] = (0, 0)) -> tuple[int, int]:
    """Generate bluetooth message id."""
    msg_id_higher_byte, msg_id_lower_byte = current_msg_id
    if msg_id_lower_byte == 255:
        if msg_id_higher_byte == 255:
            # start counting from the beginning
            return (0, 1)
        if msg_id_higher_byte == 89:
            # higher byte should never be 90
            return (msg_id_higher_byte + 2, msg_id_lower_byte)
        return (msg_id_higher_byte + 1, 0)

    else:
        if msg_id_lower_byte == 89:
            # lower byte should never be 90
            return (0, msg_id_lower_byte + 2)
        return (0, msg_id_lower_byte + 1)

def _create_command_encoding(
    cmd_id: int, cmd_mode: int, msg_id: tuple[int, int], parameters: list[int]
) -> bytearray:
    """Encode command."""
    # make sure that no parameter is 90
    sanitized_params: list[int] = list(map(lambda x: x if x != 90 else 89, parameters))

    command = bytearray(
        [cmd_id, 1, len(parameters) + 5, msg_id[0], msg_id[1], cmd_mode]
        + sanitized_params
    )

    verification_byte = _calculate_checksum(command)
    if verification_byte == 90:
        # make sure that verification byte is not 90
        new_msg_id = (msg_id[0], msg_id[1] + 1)
[... 417 lines cut ...]
│                                                                      Encode Message                                                                      │
├───────────────────────────────────────────────┬────────────┬───┬────────────────┬─────────────────┬────────────────┬──────┬───────────────────┬──────────┤
│                 Command Print                 │ Command ID │ 1 │ Command Length │ Message ID High │ Message ID Low │ Mode │     Parameters    │ Checksum │
├───────────────────────────────────────────────┼────────────┼───┼────────────────┼─────────────────┼────────────────┼──────┼───────────────────┼──────────┤
│ [165, 1, 10, 0, 218, 27, 0, 0, 0, 3, 232, 33] │    165     │ 1 │       5        │        0        │      218       │  27  │ [0, 0, 0, 3, 232] │    0     │
└───────────────────────────────────────────────┴────────────┴───┴────────────────┴─────────────────┴────────────────┴──────┴───────────────────┴──────────┘

```

```
python test2.py xx:xx:xx:xx:xx:xx --encode --send a5010a00d91b00000003de14
```

Kanal : 0 
ml    : 999,0

```
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│                                                                       Encode Message                                                                      │
├────────────────────────────────────────────────┬────────────┬───┬────────────────┬─────────────────┬────────────────┬──────┬───────────────────┬──────────┤
│                 Command Print                  │ Command ID │ 1 │ Command Length │ Message ID High │ Message ID Low │ Mode │     Parameters    │ Checksum │
├────────────────────────────────────────────────┼────────────┼───┼────────────────┼─────────────────┼────────────────┼──────┼───────────────────┼──────────┤
│ [165, 1, 10, 0, 196, 27, 0, 0, 0, 39, 15, 252] │    165     │ 1 │       5        │        0        │      196       │  27  │ [0, 0, 0, 39, 15] │    0     │
└────────────────────────────────────────────────┴────────────┴───┴────────────────┴─────────────────┴────────────────┴──────┴───────────────────┴──────────┘
```

Martin11180, 2025-09-22:
Hello
It's a shame no one has found the code to test.

I've now integrated it directly into the code, but don't want to upload it now. Here are the changes.

custom_components/chihiros/chihiros_led_control/device/base_device.py

```diff 
@property
    def rssi(self) -> int | None:
        """Get the rssi of the device."""
        if self._advertisement_data:
            return self._advertisement_data.rssi
        return None

    # Command methods

+ line added

 async def add_setting_dosing_pump(
        self,
        performance_time: Annotated[datetime, typer.Argument(formats=["%H:%M"])],
        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
        weekdays: Annotated[list[WeekdaySelect], typer.Option()] = [WeekdaySelect.everyday],
        ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
    ) -> None:
        cmd = dosingpump.create_add_auto_setting_command_dosing_pump(
            performance_time.time(),
            self.get_next_msg_id(),
            ch_id,
            encode_selected_weekdays(weekdays),
            ch_ml,
        )
        set_time_cmd  = dosingpump.create_auto_mode_dosing_pump_command_time(performance_time.time(), self.get_next_msg_id(), ch_id)
        time_cmd      = commands.create_set_time_command(self.get_next_msg_id())
        command_1_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 90, 4, 1)
        command_2_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 4)
        command_3_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 5)
        command_4_cmd = dosingpump.create_switch_to_auto_mode_dosing_pump_command(self.get_next_msg_id(), ch_id)
        await self._send_command(command_1_cmd, 3)
        await self._send_command(time_cmd, 3)
        await self._send_command(time_cmd, 3)
        await self._send_command(command_2_cmd, 3)
        await self._send_command(command_3_cmd, 3)
        await self._send_command(command_4_cmd, 3)
        await self._send_command(cmd, 3)
        await self._send_command(set_time_cmd, 3)
    
    async def set_dosing_pump_manuell_ml(
        self,
        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
        ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
    ) -> None:
        """Add an automation setting to the light."""
        cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
            self.get_next_msg_id(),
            ch_id,
            ch_ml,
        )
        await self._send_command(cmd, 3)
    
    async def enable_auto_mode_dosing_pump(
        self,
        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
      ) -> None:
        switch_cmd = dosingpump.create_switch_to_auto_mode_dosing_pump_command(self.get_next_msg_id(), ch_id)
        time_cmd   = commands.create_set_time_command(self.get_next_msg_id())
        await self._send_command(switch_cmd, 3)
        await self._send_command(time_cmd, 3)
```
[... 131 lines cut ...]

So far, manual dosing up to 25.0 ml is possible.
I need help with the code for higher doses.

--ch-id   (Channel specification for four pumps 0 - 3)
--ch-ml  (Information in ml Indication in ml for example 10 equals 1.0 ml)
Replace  XX:XX:XX:XX:XX:XXX to Mac-id Pumpe

```
chihirosctl set-dosing-pump-manuell-ml XX:XX:XX:XX:XX:XXX --ch-id 0 --ch-ml 10

```

--ch-id (Channel specification for four pumps 0 - 3)
--ch-ml (Information in ml Indication in ml for example 20 equals 2.0 ml)
--weekdays monday tuesday wednesday thursday friday saturday sunday

```

chihirosctl add-setting-dosing-pump XX:XX:XX:XX:XX:XX 1:42 --ch-id 0 --weekdays monday --ch-ml 20

```

Greet
Martin

MankiniChykan, 2025-09-22:
@Martin11180 I got your early code snippet and have been able to make it show up in the integration. I am just going to run your latest code snippet from three hours ago. Try this fork and tell be if it works [https://github.com/MankiniChykan/chihiros-led-control](https://github.com/MankiniChykan/chihiros-led-control)

MankiniChykan, 2025-09-22:
@Martin11180 From Chat GPT Nice work, Martin — thanks for sharing the details.
Your current frames put the dose amount in a single parameter byte (your ch_ml), which tops out at 0xFA = 250 → 25.0 mL (since you’re using deci-mL: 10 ⇒ 1.0 mL). To support larger doses reliably, send multiple manual-dose commands back-to-back, chunked to 25.0 mL each, with a tiny delay so the BLE stack doesn’t flap.

Below are minimal diffs that:

keep your existing wire format,

remove the 25 cap from the CLI/API,

split any dose >25.0 mL into safe chunks,

and tidy a couple bounds (channels 0–3).

1) custom_components/chihiros/chihiros_led_control/dosingpump.py

Add a constant so both sides agree on the per-command maximum.

diff --git a/custom_components/chihiros/chihiros_led_control/dosingpump.py b/custom_components/chihiros/chihiros_led_control/dosingpump.py
@@
 import datetime
 from . import commands
 from datetime import datetime
+
+# Each command carries amount in deci-mL (1 = 0.1 mL) and fits in one byte.
+# Device rejects values that resolve to 0x5A (90) or >0xFA (250).
+MAX_UNITS_PER_COMMAND = 250  # 25.0 mL
@@
 def create_add_dosing_pump_command_manuell_ml(
     msg_id: tuple[0, 0],
     ch_id: int,
     ch_ml: int,
 ) -> bytearray:
-    parameters = [ch_id, 0,  0, 0, ch_ml]
+    parameters = [ch_id, 0,  0, 0, ch_ml]
     return _create_command_encoding_dosing_pump(165, 27, msg_id, parameters)

2) custom_components/chihiros/chihiros_led_control/device/base_device.py

Chunk large manual doses; also relax the Typer limit and make channel bounds 0–3.

diff --git a/custom_components/chihiros/chihiros_led_control/device/base_device.py b/custom_components/chihiros/chihiros_led_control/device/base_device.py
@@
-from .. import commands
+from .. import commands
 from .. import dosingpump
+import asyncio
@@
-    async def set_dosing_pump_manuell_ml(
-        self,
-        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
-        ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
-    ) -> None:
-        """Add an automation setting to the light."""
-        cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
-            self.get_next_msg_id(),
-            ch_id,
-            ch_ml,
-        )
-        await self._send_command(cmd, 3)
+    async def set_dosing_pump_manuell_ml(
+        self,
+        # 4 pumps ⇒ channels 0..3
+        ch_id: Annotated[int, typer.Option(min=0, max=3)] = 0,
+        # Amount is in deci-mL (10 ⇒ 1.0 mL). Allow big values; we will chunk.
+        ch_ml: Annotated[int, typer.Option(min=0, max=9999)] = 0,
+    ) -> None:
+        """Manual dose for the dosing pump (split into ≤25.0 mL chunks)."""
+        # Convert requested amount into command-sized pieces
+        remaining = int(ch_ml)  # deci-mL
+        while remaining > 0:
[... 52 lines cut ...]
+    ch_id: Annotated[int, typer.Option(min=0, max=3)] = 0,
+    # deci-mL: 10 ⇒ 1.0 mL. Large values are OK; code will chunk.
+    ch_ml: Annotated[int, typer.Option(min=0, max=9999)] = 0,
+) -> None:
+    _run_device_func(device_address, ch_id=ch_id, ch_ml=ch_ml)

How to use (unchanged semantics)

Manual dose 50 mL on channel 0
Since --ch-ml is deci-mL, pass 500:

chihirosctl set-dosing-pump-manuell-ml XX:XX:XX:XX:XX:XX --ch-id 0 --ch-ml 500

The code sends 2×25.0 mL chunks back-to-back.

Scheduled single entry is still capped at 25.0 mL (pass up to --ch-ml 250).
If you need >25.0 mL at one time, we can extend this by creating multiple entries (e.g., 14:02, 14:03, 14:04) or, if tests confirm, by using the “count” field in your create_add_auto_setting_command_dosing_pump payload.

Why this approach?

The device appears to store the per-command amount in one byte (deci-mL), hence the natural 25.0 mL ceiling. Chunking is the simplest and most reliable way to deliver bigger doses without guessing undocumented fields.

The short await asyncio.sleep(0.25) prevents the “changed connection status while waiting for BluetoothGATTWriteResponse” error you saw under Dev Tools/ESPHome proxies.

If you later capture a single-packet payload from the app for a >25 mL dose (or a schedule that delivers >25 mL in one go), share the hex and we can add a true “wide amount” encoder.

MankiniChykan, 2025-09-22:
@Martin11180 

**New Service Call** (I increased the limit above 25mm, even though I am unsure why anyone would want to dose more than that)

[image]

**DYDOSED Automatically detected by integration**

[image]

**Controls for each channel**

[image]

[](https://github.com/MankiniChykan/chihiros-led-control) The calls are succeeding according to my logs. Be great if you can confirm they are working.

We just need a way now to read how much each dosing container still has in it (Which could be done in HA)

Be nice to have a volume remaining next to each channel

If you can sniff any other controls. i.e. setting a schedule. We may be onto a solution.

MankiniChykan, 2025-09-22:
> Hallo Haben Sie irgendwas an Setup geändert? Warum muss ich jetzt auch den HomeAssistant installieren Auf den Raspberry kommt es zu dieser Meldung
> 
> ```
> ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
> chihiros-led-control 0.1.dev132+ge90b1ec requires bleak==0.22.3, but you have bleak 1.1.1 which is incompatible.
> chihiros-led-control 0.1.dev132+ge90b1ec requires bleak_retry_connector==3.9.0, but you have bleak-retry-connector 4.4.3 which is incompatible.
> Successfully installed Jinja2-3.1.6 MarkupSafe-3.0.2 Pillow-11.1.0 PyJWT-2.10.1 PyRIC-0.1.6.3 PyTurboJPEG-1.7.5 SQLAlchemy-2.0.39 acme-3.2.0 aiodns-3.2.0 aiohappyeyeballs-2.6.1 aiohasupervisor-0.3.0 aiohttp-3.11.16 aiohttp-asyncmdnsresolver-0.1.1 aiohttp-fast-zlib-0.2.3 aiohttp_cors-0.7.0 aiosignal-1.4.0 aiozoneinfo-0.2.3 annotatedyaml-0.4.5 anyio-4.10.0 astral-2.2 async-interrupt-1.2.2 async-timeout-5.0.1 atomicwrites-homeassistant-1.4.1 attrs-25.1.0 audioop-lts-0.2.1 awesomeversion-24.6.0 bcrypt-4.2.0 bleak-1.1.1 bleak-retry-connector-4.4.3 bluetooth-adapters-2.1.1 bluetooth-auto-recovery-1.5.3 bluetooth-data-tools-1.28.2 boto3-1.40.35 botocore-1.40.35 btsocket-0.3.0 cffi-2.0.0 ciso8601-2.3.2 cronsim-2.6 cryptography-44.0.1 envs-1.4 fnv-hash-fast-1.4.0 fnvhash-0.1.0 frozenlist-1.7.0 greenlet-3.2.4 h11-0.16.0 ha-ffmpeg-3.2.2 habluetooth-5.6.4 hass-nabucasa-0.94.0 hassil-2.2.3 home-assistant-bluetooth-1.13.1 home-assistant-intents-2025.3.28 homeassistant-2025.4.4 httpcore-1.0.9 httpx-0.28.1 ifaddr-0.2.0 jmespath-1.0.1 josepy-1.15.0 lru-dict-1.3.0 mashumaro-3.16 multidict-6.6.4 mutagen-1.47.0 numpy-2.2.2 orjson-3.10.16 propcache-0.3.0 psutil-7.1.0 psutil-home-assistant-0.0.1 pyOpenSSL-25.0.0 pycares-4.11.0 pycognito-2024.5.1 pycparser-2.23 pymicro-vad-1.0.1 pyrfc3339-2.1.0 pyspeex-noise-1.0.2 python-slugify-8.0.4 requests-2.32.3 s3transfer-0.14.0 securetar-2025.2.1 sniffio-1.3.1 snitun-0.40.0 standard-aifc-3.13.0 standard-chunk-3.13.0 standard-telnetlib-3.13.0 text-unidecode-1.3 ulid-transform-1.4.0 unicode-rbnf-2.3.0 urllib3-1.26.20 uv-0.6.10 voluptuous-openapi-0.0.6 voluptuous-serialize-2.6.0 webrtc-models-0.3.0 yarl-1.18.3 zeroconf-0.146.0
> ```
> 
> Windows
> 
> ```
>            ********************************************************************************
>               Please consider removing the following classifiers in favor of a SPDX license expression:
> 
>               License :: OSI Approved :: MIT License
> 
>               See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
>               ********************************************************************************
> 
>       !!
>         self._finalize_license_expression()
>       running bdist_wheel
>       running build
>       running build_py
>       creating build\lib.win-amd64-cpython-313\lru
>       copying src\lru\__init__.py -> build\lib.win-amd64-cpython-313\lru
>       running egg_info
>       writing src\lru_dict.egg-info\PKG-INFO
>       writing dependency_links to src\lru_dict.egg-info\dependency_links.txt
>       writing requirements to src\lru_dict.egg-info\requires.txt
>       writing top-level names to src\lru_dict.egg-info\top_level.txt
>       reading manifest file 'src\lru_dict.egg-info\SOURCES.txt'
>       reading manifest template 'MANIFEST.in'
>       warning: no files found matching 'MANIFEST'
>       warning: no previously-included files matching '*.pyc' found anywhere in distribution
>       warning: no previously-included files matching '*.cache' found anywhere in distribution
>       adding license file 'LICENSE'
>       writing manifest file 'src\lru_dict.egg-info\SOURCES.txt'
>       copying src\lru\__init__.pyi -> build\lib.win-amd64-cpython-313\lru
>       copying src\lru\py.typed -> build\lib.win-amd64-cpython-313\lru
>       running build_ext
>       building 'lru._lru' extension
>       error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
>       [end of output]
> 
>   note: This error originates from a subprocess, and is likely not a problem with pip.
>   ERROR: Failed building wheel for lru-dict
> Failed to build ciso8601 lru-dict
> error: failed-wheel-build-for-install
> 
> × Failed to build installable wheels for some pyproject.toml based projects
> ╰─> ciso8601, lru-dict
> ```
> 
> Können Sie so ein Code wie zum Beispiel meinen, zum Debugen auch bei Ihnen einbauen und auch zum testen der Daten per ctl übergabe Command ID Mode [Parameters übergabe ]
> 
> Wie müss ich den code jetzt aufrufen per ctl?
> 
> Greet Martin

You should be able to still download and test it directly in HA. I also changed these files trying to get the codespaces up and running.

`setup.cfg`
```
# Setuptools v62.3 doesn't support editable installs with just 'pyproject.toml' (PEP 660).
# Keep this file until it does!

[metadata]
url = https://github.com/TheMicDiet/chihiros-led-control
[... 43 lines cut ...]
]
keywords = ["chihiros", "bluetooth", "lib"]

requires-python = ">=3.12"

[project.urls]
"Source Code" = "https://github.com/TheMicDiet/chihiros-led-control"
"Bug Reports" = "https://github.com/TheMicDiet/chihiros-led-control/issues"
"Home-page"   = "https://github.com/TheMicDiet/chihiros-led-control"

[tool.setuptools]
platforms = ["any"]
include-package-data = true

[tool.setuptools.packages.find]
include = [
  "custom_components*",
]

[tool.setuptools_scm]

[project.scripts]
chihirosctl = "custom_components.chihiros.chihiros_led_control.chihirosctl:app"

```

Martin11180, 2025-09-22:
I use openhab and would like to call it via chihirosctl

MankiniChykan, 2025-09-22:
OK... I rely heavily of Chat GPT when putting these things together (Unfortunately I know enough to break things and a basic amount to steer Chat GPT in the right direction to fix it.). I can try help you with that. I am no programmer self taught.

MankiniChykan, 2025-09-22:
This may help fix it.

Hi Martin! 👋
kurze, ehrliche Antworten + konkrete Fixes:

---

## Warum plötzlich Home Assistant installiert wird / Versionskonflikte

Du hast die Python-Bibliothek **`chihiros-led-control`** (mit CLI) in **demselben** Python-Environment installiert, in dem auch **Home Assistant** gelandet ist.
Dadurch prallen Abhängigkeiten aufeinander:

* Deine Lib pinnt `bleak==0.22.3` und `bleak_retry_connector==3.9.0`.
* Dein HA (2025.4.x) bringt/erwartet **neue** Versionen (`bleak 1.1.1`, `bleak-retry-connector 4.4.3`).

In **einem** venv kann pip das nicht sauber auflösen → Warnungen & Fehler.

### Saubere Lösung

* **Zwei getrennte Umgebungen**:

  1. **HA** (HA OS / Container / eigenes venv) – dort niemals selbst `bleak` pinnen.
  2. **CLI/Debug** (eigenes venv nur für das Tool).

#### Beispiel auf Raspberry Pi OS

```bash
# 1) CLI-venv für dein Tool
python3 -m venv ~/chyki-cli
source ~/chyki-cli/bin/activate
pip install -U pip

# Minimal, kompatibel zu deiner Lib:
pip install "bleak==0.22.3" "bleak-retry-connector==3.9.0" "typer[all]==0.15.2"

# Falls du dein Repo lokal nutzen willst:
# (Passe den Pfad an – dort, wo das Paket chihiros_led_control liegt)
pip install -e ~/pfad/zu/deinem/repo
```

**Wichtig:** Dieses CLI-venv **nicht** mit Home Assistant mischen.
HA selbst bitte getrennt laufen lassen (HA OS/Container oder eigenes venv).

### Windows-Fehler (MSVC)

Beim Versuch, **Home Assistant** per pip unter Windows zu bauen, scheitern native Wheels (`lru-dict`, `ciso8601`) → MSVC Build Tools nötig.
Empfehlung: Für HA **WSL/Container** nutzen, oder alles unter Linux/RPi. Für dein **CLI** reicht das kleine venv wie oben.

---

## HACS-404 („e90b1ec.zip“)

HACS hat versucht, eine **nicht existierende** Branch/Ref (`e90b1ec`) als Zip zu laden → 404.
Fix:

* In HACS das Repo **löschen** und **neu** hinzufügen (Branch `main` oder ein **Tag**).
* Oder in HACS unter „Version“ einen gültigen **Tag/Commit** auswählen.

---

## Dein Service-Call (Dev Tools) schlägt fehl („expected float for ml“)

Das kommt, wenn `ml` gar nicht gesetzt ist oder als `null` reinkommt.
Beispielpayloads, die **funktionieren**:

**Variante A – über `address`:**

```json
{
  "address": "D2:03:E0:FE:FC:BC",
  "channel": 1,
  "ml": 5.0
}
```

**Variante B – über `device_id`:**

```json
{
  "device_id": "abcd1234efgh5678ijkl9012mnop3456",
  "channel": 1,
  "ml": 5.0
}
```

Hinweise:

* `channel` ist im Service **1..4** (nicht 0..3).
* `ml` muss **float** sein → `5.0` statt `5` (wir können das später auf `vol.Coerce(float)` umstellen, dann geht auch `5`).

---

## „ctl“ (CLI) – so rufst du es auf

Wenn dein CLI richtig installiert ist (getrenntes venv wie oben) und dein Typer-Command `set-dosing-pump-manuell-ml` existiert:

```bash
# 50 mL auf Kanal 0 (deci-mL ⇒ 500)
chihirosctl set-dosing-pump-manuell-ml AA:BB:CC:DD:EE:FF --ch-id 0 --ch-ml 500
```

Für geplante Dosen bis 25.0 mL:

```bash
chihirosctl add-setting-dosing-pump AA:BB:CC:DD:EE:FF 14:02 --ch-id 0 --weekdays monday --ch-ml 200
```

> Deine Firmware akzeptiert pro Frame max. **25.0 mL** (250 deci-mL). Für größere **manuelle** Dosen splitten wir das intern in 25-mL-Häppchen mit Mini-Pause – das ist in deinem HA-Service bereits drin.

---

## Wunsch: „raw command“ per CLI (Command-ID, Mode, Parameter)

Ja – kannst du sehr leicht einbauen. So hast du eine **universelle** Rohschnittstelle zum Debuggen.

### 1) In `custom_components/chihiros/chihiros_led_control/device/base_device.py`

```python
from .. import commands
from .. import dosingpump

# Neu:
from typing import List

# ...

    async def raw_dosing_pump(
        self,
        cmd_id: int,
        mode: int,
        params: List[int],
        repeats: int = 3,
    ) -> None:
        """
        Sendet einen Roh-Frame im Dosing-Pump-Format:
        cmd_id, mode, [params...]  -> automatisch korrekt kodiert + checksum.
        """
        pkt = dosingpump._create_command_encoding_dosing_pump(
            cmd_id, mode, self.get_next_msg_id(), params
        )
        await self._send_command(pkt, repeats)
```

### 2) In `custom_components/chihiros/chihiros_led_control/chihirosctl.py`

```python
from typing import List

# ...

@app.command()
def raw_dosing_pump(
    device_address: str,
    cmd_id: int,
    mode: int,
    params: List[int] = typer.Argument(..., help="Parameter als Liste, z.B. 0 0 0 0 10"),
    repeats: int = typer.Option(3, help="Sende-Wiederholungen"),
) -> None:
    _run_device_func(
        device_address,
        cmd_id=cmd_id,
        mode=mode,
        params=params,
        repeats=repeats,
    )
```

### Aufrufbeispiele

```bash
# Entspricht deinem manuellen 1.0 mL Beispiel (10 deci-mL):
chihirosctl raw-dosing-pump AA:BB:CC:DD:EE:FF --cmd-id 165 --mode 27 --params 0 0 0 0 10

# Auto-Setting Zeit (Beispiel):
chihirosctl raw-dosing-pump AA:BB:CC:DD:EE:FF --cmd-id 165 --mode 21 --params 0 0 14 02 0 0

# Auto-Mode Umschalten (Beispiel):
chihirosctl raw-dosing-pump AA:BB:CC:DD:EE:FF --cmd-id 165 --mode 32 --params 0 0 1
```

> Vorteil: Du kannst **jede** Frame-Kombination testen, ohne das High-Level API anzupassen.

---

## Noch zwei Quick-Tips

* Wenn ein **Button** aus dem Gerät sofort funktioniert, der **Dev-Tools Service** aber meckert, ist es fast immer der Payload (fehlendes `ml`, falscher `channel`, falscher Target-Typ). Nimm die JSON-Beispiele oben 1:1.
* Siehst du „BleakClient.connect() called without bleak-retry-connector“, ist das nur ein Hinweis. In HA verwenden wir bereits robuste Verbindungen; bei rohen Tests ggf. kleine `await asyncio.sleep(0.25)` Pausen zwischen Frames lassen (hast du in der HA-Service-Implementierung schon drin).

---

Wenn du magst, schicke mir **genau den Dev-Tools JSON** (ohne MAC/IDs) der fehlschlägt – dann sag ich dir, welche Kleinigkeit fehlt. 💪

Martin11180, 2025-09-22:
Hallo
Können wir hier nicht auf deutsch schreiben?

Können Sie mein Code snippel angepasst an ihren Code  einfügen , das ich nicht immer wieder einbauen muss und es über ctl geht?
Auch  der Debug code  raw command und an besten sowas wie mein Code zum encoden  von hex to Message, wie oben
Es gibt ja noch einige Sachen , die man debuggen kann und Habe ja auch noch die led chihiros wrgb 1000 wo noch nicht alles läuft
Ich bin die ganze Zeit dabei den  homeassistant mal zum laufen zum bringen auf Rasberry  und ihr code einzubauen
Mit den Kanal ich bin der auffassung , das meiner bei 0 startet also 1 gleich 0
Ich weiss nicht ob Sie es gesehen haben in meinen Beitrag ml: 25,5 ist der Parameter noch in der letzten Spalte [0, 0, 0, 0, 255]
ab 25,6 [0, 0, 0, 1 0] also die letzten beiden 
ml : 999,0  [0, 0, 0, 3, 232]

Hello
Can't we write in German here?

Can you insert my code snippet adapted to your code so that I don't have to re-insert it and it works via CTL?
The debug code is also a raw command, and ideally something like my code for encoding hex to message, as above.
There are still a few things that can be debugged, and I also still have the Chihiros WRGB 1000 LED, where not everything is working yet.
I'm currently working on getting Home Assistant running on Raspberry and integrating its code.
Regarding the channel, I believe mine starts at 0, so 1 equals 0.
I don't know if you saw it in my post ml: 25.5, the parameter is still in the last column [0, 0, 0, 0, 255]
from 25.6 [0, 0, 0, 1 0], so the last two
ml: 999.0 [0, 0, 0, 3, 232]

MankiniChykan, 2025-09-22:
> Hello It's a shame no one has found the code to test.
> 
> I've now integrated it directly into the code, but don't want to upload it now. Here are the changes.
> 
> custom_components/chihiros/chihiros_led_control/device/base_device.py
> 
> from .. import commands
> 
> + line added
> 
> from .. import dosingpump
> 
> next 
> 
> @property
>     def rssi(self) -> int | None:
>         """Get the rssi of the device."""
>         if self._advertisement_data:
>             return self._advertisement_data.rssi
>         return None
> 
>     # Command methods
> 
> + line added
>    async def add_setting_dosing_pump(
>         self,
>         performance_time: Annotated[datetime, typer.Argument(formats=["%H:%M"])],
>         ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
>         weekdays: Annotated[list[WeekdaySelect], typer.Option()] = [WeekdaySelect.everyday],
>         ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
>     ) -> None:
>         cmd = dosingpump.create_add_auto_setting_command_dosing_pump(
>             performance_time.time(),
>             self.get_next_msg_id(),
>             ch_id,
>             encode_selected_weekdays(weekdays),
>             ch_ml,
>         )
>         set_time_cmd  = dosingpump.create_auto_mode_dosing_pump_command_time(performance_time.time(), self.get_next_msg_id(), ch_id)
>         time_cmd      = commands.create_set_time_command(self.get_next_msg_id())
>         command_1_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 90, 4, 1)
>         command_2_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 4)
>         command_3_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 5)
>         command_4_cmd = dosingpump.create_switch_to_auto_mode_dosing_pump_command(self.get_next_msg_id(), ch_id)
>         await self._send_command(command_1_cmd, 3)
>         await self._send_command(time_cmd, 3)
>         await self._send_command(time_cmd, 3)
>         await self._send_command(command_2_cmd, 3)
>         await self._send_command(command_3_cmd, 3)
>         await self._send_command(command_4_cmd, 3)
>         await self._send_command(cmd, 3)
>         await self._send_command(set_time_cmd, 3)
>     
>     async def set_dosing_pump_manuell_ml(
>         self,
>         ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
>         ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
>     ) -> None:
>         """Add an automation setting to the light."""
>         cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
>             self.get_next_msg_id(),
>             ch_id,
>             ch_ml,
>         )
>         await self._send_command(cmd, 3)
>     
>     async def enable_auto_mode_dosing_pump(
>         self,
>         ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
>       ) -> None:
[... 120 lines cut ...]
> def create_switch_to_auto_mode_dosing_pump_command(
>     msg_id: tuple[int, int],
>     channel_id
>     ) -> bytearray:
>     """Create switch auto setting command."""
>     return _create_command_encoding_dosing_pump(165, 32, msg_id, [channel_id, 0, 1])
> 
> So far, manual dosing up to 25.0 ml is possible. I need help with the code for higher doses.
> 
> --ch-id (Channel specification for four pumps 0 - 3) --ch-ml (Information in ml Indication in ml for example 10 equals 1.0 ml) Replace XX:XX:XX:XX:XX:XXX to Mac-id Pumpe
> 
> ```
> chihirosctl set-dosing-pump-manuell-ml XX:XX:XX:XX:XX:XXX --ch-id 0 --ch-ml 10
> ```
> 
> --ch-id (Channel specification for four pumps 0 - 3) --ch-ml (Information in ml Indication in ml for example 20 equals 2.0 ml) --weekdays monday tuesday wednesday thursday friday saturday sunday
> 
> ```
> 
> chihirosctl add-setting-dosing-pump XX:XX:XX:XX:XX:XX 1:42 --ch-id 0 --weekdays monday --ch-ml 20
> ```
> 
> Greet Martin

Is this your base code?

MankiniChykan, 2025-09-22:
Here’s a **minimal, surgical diff** that should allow doses **above 25 ml** by automatically splitting the request into **25.0 ml “buckets”** (i.e., 250 tenths-of-ml units) and sending multiple commands back-to-back. It keeps all current CLI flags and semantics the same (e.g., `--ch-ml 10` = **1.0 ml**).

I’ve limited the change surface to three places:

1. lift the CLI caps so larger values are allowed,
2. add a tiny chunking loop in the device methods,
3. define one constant to keep “bucket size” in one place.

---

### 1) `custom_components/chihiros/chihiros_led_control/dosingpump.py`

Add a single constant (and optionally export it) so there’s one source of truth for the bucket size.

```diff
@@
 import datetime
 from . import commands
 from datetime import datetime
 
+# Each device command can safely carry up to 25.0 ml (i.e. 250 tenths-of-ml units).
+# Keep this centralized so we can tweak later without touching call sites.
+MAX_BUCKET_UNITS = 250  # 250 units == 25.0 ml when 1 unit == 0.1 ml
+
 def _create_command_encoding_dosing_pump(
     cmd_id: int, cmd_mode: int, msg_id: tuple[int, int], parameters: list[int]
 ) -> bytearray:
```

*No other changes required in this file.*
(We’ll reference `MAX_BUCKET_UNITS` from the device code.)

---

### 2) `custom_components/chihiros/chihiros_led_control/device/base_device.py`

Lift the option caps and add **tiny chunking** in both the manual and auto flows. The command format stays identical; we just send **N** commands when the requested amount is larger than a single bucket.

```diff
@@
 from .. import commands
+from .. import dosingpump
@@
     # Command methods
-   async def add_setting_dosing_pump(
+   async def add_setting_dosing_pump(
         self,
         performance_time: Annotated[datetime, typer.Argument(formats=["%H:%M"])],
-        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
-        weekdays: Annotated[list[WeekdaySelect], typer.Option()] = [WeekdaySelect.everyday],
-        ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
+        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
+        weekdays: Annotated[list[WeekdaySelect], typer.Option()] = [WeekdaySelect.everyday],
+        # Accept larger totals; we will split into buckets of MAX_BUCKET_UNITS internally.
+        # Units are 0.1 ml; e.g., 10 == 1.0 ml, 250 == 25.0 ml, 1000 == 100.0 ml.
+        ch_ml: Annotated[int, typer.Option(min=0)] = 0,
     ) -> None:
         cmd = dosingpump.create_add_auto_setting_command_dosing_pump(
             performance_time.time(),
             self.get_next_msg_id(),
             ch_id,
             encode_selected_weekdays(weekdays),
-            ch_ml,
+            min(ch_ml, dosingpump.MAX_BUCKET_UNITS),
         )
         set_time_cmd  = dosingpump.create_auto_mode_dosing_pump_command_time(performance_time.time(), self.get_next_msg_id(), ch_id)
         time_cmd      = commands.create_set_time_command(self.get_next_msg_id())
         command_1_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 90, 4, 1)
         command_2_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 4)
         command_3_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 5)
[... 101 lines cut ...]
  chihirosctl set-dosing-pump-manuell-ml XX:XX:XX:XX:XX:XXX --ch-id 0 --ch-ml 600
  ```

  The device method sends **3 commands**: `25.0 + 25.0 + 10.0 ml`.

* **Scheduled 80 ml at 01:42 Mondays** on channel 0:

  ```bash
  chihirosctl add-setting-dosing-pump XX:XX:XX:XX:XX:XXX 1:42 --ch-id 0 --weekdays monday --ch-ml 800
  ```

  The device method attempts to create **multiple settings at the same time** to sum to 80.0 ml.
  If your firmware only stores **one** entry per time, change that loop to **only** chunk manual doses (leave the scheduled one single-shot at `min(ch_ml, 25.0 ml)`).

---

## Notes & safety

* We explicitly **don’t** change the wire format. Every “bucket” uses the same command shape the device already understands.
* `MAX_BUCKET_UNITS` (250) centralizes the per-command ceiling. If field testing shows a different hard limit, tweak that one number.
* If the device enforces a **rate limit**, add a short `asyncio.sleep(0.1)` inside the while-loops.

If you want me to flip the scheduled path to “single-entry only” (and keep chunking just for manual), say the word and I’ll give you the tiny diff (it’s a 3-liner).

---

Martin11180, 2025-09-22:
I don't understand the translation.
You speak German, right?
Can I contact you by email?

MankiniChykan, 2025-09-22:
No Mate I am Australian. I will translate the last diff to German for you.

MankiniChykan, 2025-09-22:
Try this. 

Klar—hier ist ein **einzelner, direkt einfügbarer Unified-Diff**, der alles umsetzt, was wir besprochen haben: Unterstützung für **0,1-ml-Schritte von 0,0 → 999,9 ml**, Aufteilung in **≤ 25,0-ml-Buckets** sowie **automatisches Staffeln** geplanter Buckets im Minutentakt, damit Firmwares, die pro HH\:MM nur einen Eintrag behalten, frühere Einträge nicht überschreiben.

> Die Pfade sind exakt wie von dir angegeben. Der Patch:
>
> * **fügt** `dosingpump.py` **hinzu**
> * **aktualisiert** `base_device.py`, um sie zu importieren, Limits anzuheben, zu “chunken” & zu staffeln
> * **aktualisiert** `chihirosctl.py`, um die CLI-Limits anzuheben und die neuen Befehle hinzuzufügen

---

```diff
diff --git a/custom_components/chihiros/chihiros_led_control/device/base_device.py b/custom_components/chihiros/chihiros_led_control/device/base_device.py
index 0000000..0000000 100644
--- a/custom_components/chihiros/chihiros_led_control/device/base_device.py
+++ b/custom_components/chihiros/chihiros_led_control/device/base_device.py
@@ -1,6 +1,8 @@
-from .. import commands
+from .. import commands
+from .. import dosingpump
+from datetime import datetime, timedelta
 from typing_extensions import Annotated
 import typer
@@
     def rssi(self) -> int | None:
         """Get the rssi of the device."""
         if self._advertisement_data:
             return self._advertisement_data.rssi
         return None
@@
-    # Command methods
+    # Command methods
+
+    def _add_minutes(self, t: datetime.time, delta_min: int) -> datetime.time:
+        """Return t + delta_min minutes (HH:MM), wrapping hours if needed."""
+        anchor = datetime(2000, 1, 1, t.hour, t.minute)
+        out = anchor + timedelta(minutes=delta_min)
+        return out.time()
 
-    async def add_setting_dosing_pump(
+    async def add_setting_dosing_pump(
         self,
         performance_time: Annotated[datetime, typer.Argument(formats=["%H:%M"])],
         ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
         weekdays: Annotated[list[WeekdaySelect], typer.Option()] = [WeekdaySelect.everyday],
-        ch_ml: Annotated[int, typer.Option(max=25, min=0)] = 0,
+        # 0.1 ml units (1 == 0.1 ml). Allow up to 9999 == 999.9 ml.
+        ch_ml: Annotated[int, typer.Option(min=0, max=9999)] = 0,
     ) -> None:
-        cmd = dosingpump.create_add_auto_setting_command_dosing_pump(
-            performance_time.time(),
-            self.get_next_msg_id(),
-            ch_id,
-            encode_selected_weekdays(weekdays),
-            ch_ml,
-        )
-        set_time_cmd  = dosingpump.create_auto_mode_dosing_pump_command_time(performance_time.time(), self.get_next_msg_id(), ch_id)
+        # Prelude & mode/time sync (unchanged semantics)
+        set_time_cmd0 = dosingpump.create_auto_mode_dosing_pump_command_time(
+            performance_time.time(), self.get_next_msg_id(), ch_id
+        )
         time_cmd      = commands.create_set_time_command(self.get_next_msg_id())
         command_1_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 90, 4, 1)
         command_2_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 4)
         command_3_cmd = dosingpump.create_order_confirmation(self.get_next_msg_id(), 165, 4, 5)
         command_4_cmd = dosingpump.create_switch_to_auto_mode_dosing_pump_command(self.get_next_msg_id(), ch_id)
         await self._send_command(command_1_cmd, 3)
         await self._send_command(time_cmd, 3)
         await self._send_command(time_cmd, 3)
[... 190 lines cut ...]
+) -> bytearray:
+    """Create 'switch to auto mode' for the dosing pump."""
+    return _create_command_encoding_dosing_pump(165, 32, msg_id, [channel_id, 0, 1])
```

---

### Was dieser Patch liefert

* **Manuelle Dosierung** (`set-dosing-pump-manuell-ml`): Beliebige Gesamtmenge **0,0–999,9 ml** in **0,1-ml-Schritten**, versendet in Buckets ≤ 25,0 ml.
* **Geplante Dosierung** (`add-setting-dosing-pump`): Gleiches Spektrum und gleiche Auflösung; zusätzliche Buckets werden **automatisch im Minutentakt** ab der gewünschten `HH:MM` gestaffelt, um ein Firmware-Verhalten “nur ein Eintrag pro Minute” nicht zu überschreiben.

Wenn du später bestätigst, dass deine Firmware **Einträge zur selben Minute stapelt**, kannst du das Staffeln entfernen, indem du `t_i = self._add_minutes(performance_time.time(), minute_offset)` einfach durch `performance_time.time()` ersetzt.

---

Soll ich dir das als `.patch` bauen, damit du es mit `git am` einspielen kannst? Mach ich gern!

```bash
# Dev release
npm run build && npm run release:dev -- 1.0.327-dev.0

# Production release
npm run build && npm run release -- 1.0.327
```

MankiniChykan, 2025-09-24:
I will give it a go when I am near a computer. Just traveling home 3000km from one end of Australia to the other. I will be able to finally test this with my doser. Did you get it loaded into Home Assistant?

MankiniChykan, 2025-09-24:
Perfect — here are the ready-to-paste additions to enable reading:

current auto dosing settings; and

container total/remaining volume.

They plug straight into the drop-in files I gave you.

---

1) Add (or keep) these in dosingpump.py

If you already have these helpers/consts from earlier, you can skip this block. Otherwise paste it near the bottom of dosingpump.py (below the write command builders).

# ────────────────────────────────────────────────────────────────
# READ scaffolds (fill opcodes once sniffed)
# ────────────────────────────────────────────────────────────────
# TODO: set the real values after sniffing the OEM app:
QUERY_AUTO_SETTINGS_CMD_ID = 165
QUERY_AUTO_SETTINGS_MODE   = 22

QUERY_CONTAINER_STATUS_CMD_ID = 165
QUERY_CONTAINER_STATUS_MODE   = 23

def create_query_auto_settings_command(
    msg_id: tuple[int, int],
    channel_id: int | None = None,
) -> bytearray:
    """Build a 'query auto-dosing settings' command (single channel or all)."""
    params = [255] if channel_id is None else [channel_id & 0xFF]
    return _create_command_encoding_dosing_pump(
        QUERY_AUTO_SETTINGS_CMD_ID, QUERY_AUTO_SETTINGS_MODE, msg_id, params
    )

def create_query_container_status_command(
    msg_id: tuple[int, int],
    channel_id: int | None = None,
) -> bytearray:
    """Build a 'query container total/remaining' command (single or all channels)."""
    params = [255] if channel_id is None else [channel_id & 0xFF]
    return _create_command_encoding_dosing_pump(
        QUERY_CONTAINER_STATUS_CMD_ID, QUERY_CONTAINER_STATUS_MODE, msg_id, params
    )

# ────────────────────────────────────────────────────────────────
# Best-effort parsers (adjust once you see real payloads)
# ────────────────────────────────────────────────────────────────
def try_parse_auto_settings(payload: bytes) -> list[dict] | None:
    """
    Heuristic: scan 6-byte windows like the write layout:
      [channel, weekdays_mask, b2, b3, b4, amount_units]
    """
    out: list[dict] = []
    n = len(payload)
    for i in range(0, max(0, n - 5)):
        chunk = payload[i:i + 6]
        if len(chunk) < 6:
            break
        ch_id, weekdays, _b2, _b3, _b4, amount = chunk
        if 0 <= ch_id <= 3 and 0 <= amount <= MAX_BUCKET_UNITS:
            out.append({
                "channel": ch_id,
                "weekdays_mask": weekdays & 0xFF,
                "amount_units": amount,
                "amount_ml": amount / 10.0,
                "offset": i,
            })
    return out or None

def _le16(lo: int, hi: int) -> int:
[... 155 lines cut ...]
    """
    _run_device_func(
        device_address,
        ch_id=ch_id,
        timeout_s=timeout_s,
        func_name="read_dosing_container_status",
    )

---

Quick test examples

# Read all channels' auto settings (after you set the real QUERY opcodes)
chihirosctl read-dosing-auto AA:BB:CC:DD:EE:FF

# Read only channel 2
chihirosctl read-dosing-auto AA:BB:CC:DD:EE:FF --ch-id 2

# Read container totals/remaining for all
chihirosctl read-dosing-container AA:BB:CC:DD:EE:FF

# Only channel 0
chihirosctl read-dosing-container AA:BB:CC:DD:EE:FF --ch-id 0

If you paste a sample raw response hex from either read, I’ll tighten the parser to the exact field layout (usually a <10-line tweak).

MankiniChykan, 2025-09-24:
Great catch — your sample frame shows you’re parsing a notification (RX) frame, not the command (TX) frame. The two use different “length” semantics:

TX (what we send): length = param_len + 5
Header is [cmd_id, 1, length, msg_hi, msg_lo, mode, …params…, checksum]

RX (what we receive / notification): in your sample length = param_len + 2
(same header layout, but the meaning of length differs)

Your example:

[91, 1, 10, 0, 1, 34,  0,92,  0,71,  0,107,  0,138,  51]
 ^    ^   ^   ^  ^  ^   \________ params (8B) _______/   ^
ID    v   L   MH ML MO                                  checksum

L = 10 → param_len = 10 - 2 = 8  ✅

Those 8 param bytes are four little-endian u16 values:
0x005C=92 → 9.2 ml, 0x0047=71 → 7.1 ml, 0x006B=107 → 10.7 ml, 0x008A=138 → 13.8 ml (channels 0..3 in order).

Fix your bytes_encode parser

Use a dual rule (auto-detect RX vs TX) and always take the checksum as the last byte. Also avoid negative slicing; compute start/end explicitly.

Drop-in replacement for the parsing core:

# Fields per frame:
# [cmd_id, 1, length_field, msg_hi, msg_lo, mode, <params...>, checksum]
cmd_id = value_bytes[0] if len(value_bytes) >= 1 else "????"
proto_ver = value_bytes[1] if len(value_bytes) >= 2 else "????"
length_field = value_bytes[2] if len(value_bytes) >= 3 else None
msg_hi = value_bytes[3] if len(value_bytes) >= 4 else "????"
msg_lo = value_bytes[4] if len(value_bytes) >= 5 else "????"
mode = value_bytes[5] if len(value_bytes) >= 6 else "????"

# Compute parameter length
param_len_from_total = max(0, len(value_bytes) - 7)  # total - (6 header + 1 checksum)

param_len = None
if isinstance(length_field, int):
    if cmd_id == 0x5B or cmd_id == 91:
        # RX/notification observed in your capture: length = params + 2
        param_len = max(0, length_field - 2)
    else:
        # Default to TX rule: length = params + 5
        param_len = max(0, length_field - 5)

# If mismatch, prefer the total-based value (more robust)
if param_len is None or param_len != param_len_from_total:
    param_len = param_len_from_total

# Slice params safely
params_start = 6
params_end = min(len(value_bytes) - 1, params_start + param_len)  # up to before checksum
params_list = [int(b) for b in value_bytes[params_start:params_end]]

# Checksum is always the last byte
checksum = value_bytes[-1] if len(value_bytes) >= 1 else "????"

Optional: decode RX params as four channel ml values

If cmd_id indicates an RX frame and param_len == 8, show a friendly decode:

def _le16(lo, hi): return (lo & 0xFF) | ((hi & 0xFF) << 8)

pretty_decode = None
if (cmd_id == 0x5B or cmd_id == 91) and len(params_list) == 8:
    vals = []
    for i in range(0, 8, 2):
        u = _le16(params_list[i], params_list[i+1])
        vals.append(u / 10.0)  # 0.1 ml units
    pretty_decode = {
        "ch0_ml": vals[0], "ch1_ml": vals[1], "ch2_ml": vals[2], "ch3_ml": vals[3]
    }

Then include pretty_decode in your table (if present), or print an extra line:

if pretty_decode:
    typer.echo(
        f"Decoded (ml): ch0={pretty_decode['ch0_ml']:.1f}, "
        f"ch1={pretty_decode['ch1_ml']:.1f}, "
        f"ch2={pretty_decode['ch2_ml']:.1f}, "
        f"ch3={pretty_decode['ch3_ml']:.1f}"
    )

Why your old slice broke

With command_length = int(value_bytes[2]) - 5 (TX rule), your RX frame reported too few params, so the slice [-command_length-1:-1] grabbed the wrong region and even mis-identified the checksum as 71 instead of the actual final byte 51.

Switching to explicit start/end indices and using dual length rules fixes both TX and RX cases.

If you share another RX sample (different cmd_id/mode), I can extend the auto-detection table so bytes_encode prints context (e.g., “daily consumption per channel”).

MankiniChykan, 2025-09-24:
Sorry mate working from a tablet on a plane. When I am sitting down tomorrow I will be able to help a bit better.

MankiniChykan, 2025-09-24:
Ah yes. I have just been giving you fixes for yours. I will work on putting these additions to my code tomorrow. If you have my fork of this integration installed you will be able to just download the updates straight to home assistant. 

This is hopefully the response you wanted?

Here’s a single, copy-pasteable unified diff covering all touched files. It:

adds dosingpump.py

updates device/base_device.py (chunking to ≤25.0 ml, 0.1 ml precision up to 999.9 ml, minute-stagger for scheduled multi-buckets, raw helper)

updates chihirosctl.py (robust raw-dosing-pump with --params, smarter bytes-encode that supports RX/TX frames, plus read CLIs)

> Paths match your layout: custom_components/chihiros/chihiros_led_control/...

---

diff --git a/custom_components/chihiros/chihiros_led_control/device/base_device.py b/custom_components/chihiros/chihiros_led_control/device/base_device.py
index 0000000..0000001 100644
--- a/custom_components/chihiros/chihiros_led_control/device/base_device.py
+++ b/custom_components/chihiros/chihiros_led_control/device/base_device.py
@@ -1,6 +1,12 @@
-from .. import commands
+from .. import commands
+from .. import dosingpump
+import asyncio
+import typer
+from typing import List
+from typing_extensions import Annotated
+from datetime import datetime, timedelta
 from typing_extensions import Annotated
 import typer
@@
     def rssi(self) -> int | None:
         """Get the rssi of the device."""
         if self._advertisement_data:
             return self._advertisement_data.rssi
         return None
@@
-    # Command methods
+    # Command methods
+
+    def _add_minutes(self, t: datetime.time, delta_min: int) -> datetime.time:
+        """Return t + delta_min minutes (HH:MM), wrapping hours if needed."""
+        anchor = datetime(2000, 1, 1, t.hour, t.minute)
+        out = anchor + timedelta(minutes=delta_min)
+        return out.time()
+
+    async def raw_dosing_pump(
+        self,
+        cmd_id: Annotated[int, typer.Option()] = 0,
+        # do NOT cap at 3; valid modes include 21, 27, 32, ...
+        mode: Annotated[int, typer.Option(min=0)] = 0,
+        params: Annotated[List[int], typer.Argument(help="Parameter-Liste, z.B. 0 0 14 02 0 0")] = None,
+        repeats: int = 3,
+    ) -> None:
+        """
+        Sendet einen Roh-Frame im Dosing-Pump-Format: cmd_id, mode, [params...] – inkl. Checksumme.
+        """
+        p = params or []
+        pkt = dosingpump._create_command_encoding_dosing_pump(  # type: ignore[attr-defined]
+            cmd_id, mode, self.get_next_msg_id(), p
+        )
+        await self._send_command(pkt, repeats)
 
-    async def add_setting_dosing_pump(
+    async def add_setting_dosing_pump(
         self,
         performance_time: Annotated[datetime, typer.Argument(formats=["%H:%M"])],
-        ch_id: Annotated[int, typer.Option(max=4, min=0)] = 0,
+        ch_id: Annotated[int, typer.Option(max=3, min=0)] = 0,
[... 467 lines cut ...]
+            continue
+        total_u = _le16(payload[i + 1], payload[i + 2])
+        remain_u = _le16(payload[i + 3], payload[i + 4])
+        if 0 <= remain_u <= total_u <= 200_000:
+            out.append({
+                "channel": ch,
+                "total_units": total_u,
+                "total_ml": total_u / 10.0,
+                "remaining_units": remain_u,
+                "remaining_ml": remain_u / 10.0,
+                "offset": i,
+            })
+    return out or None

---

Notes

In bytes_encode, we now handle RX (cmd_id=0x5B) by using param_len = length_field - 2 and we also cross-check with the total frame size; this matches your sniffed notifications and cleanly decodes the 4×u16 daily totals (0.1 ml units) for channels 0–3.

The read helpers in base_device.py require you to set the correct QUERY_* opcode/mode constants in dosingpump.py once you sniff them from the OEM app.

If your project already has class scaffolding around BaseDevice, merge the method bodies rather than replacing the whole class.

If you want this emitted as a .patch file you can apply with git am, I can package it exactly as such.

MankiniChykan, 2025-09-24:
Good start! Here’s a clean, drop-in version with fixes:

from __future__ import annotations (correct future import)

Proper indentation

Type hints

Atomic writes (avoid corrupting JSON on crash)

Basic validation and safe uppercasing of MAC

Save as: custom_components/chihiros/chihiros_led_control/storage_containers.py

from __future__ import annotations

import json
from pathlib import Path
from typing import Dict, Any
import tempfile
import os

_STORE = Path.home() / ".chihiros" / "containers.json"
_STORE.parent.mkdir(parents=True, exist_ok=True)

def _atomic_write(path: Path, text: str) -> None:
    """Write text atomically to avoid truncation/corruption."""
    with tempfile.NamedTemporaryFile("w", delete=False, dir=str(path.parent), encoding="utf-8") as tmp:
        tmp.write(text)
        tmp_path = Path(tmp.name)
    os.replace(tmp_path, path)

def _load() -> Dict[str, Any]:
    if _STORE.exists():
        try:
            return json.loads(_STORE.read_text(encoding="utf-8"))
        except Exception:
            # Corrupt or unreadable; keep user safe and start fresh.
            return {}
    return {}

def _save(data: Dict[str, Any]) -> None:
    _atomic_write(_STORE, json.dumps(data, indent=2, sort_keys=True))

def _key(mac: str) -> str:
    # Normalize MAC-ish identifiers; keep only hex and separators uppercased
    return str(mac).strip().upper()

def _ensure_channel(data: Dict[str, Any], mac: str, ch: int) -> Dict[str, Any]:
    dev = data.setdefault(_key(mac), {})
    return dev.setdefault(str(int(ch)), {})

def get_capacity(mac: str, ch: int) -> float:
    data = _load()
    return float(data.get(_key(mac), {}).get(str(int(ch)), {}).get("capacity_ml", 0.0))

def get_remaining(mac: str, ch: int) -> float:
    data = _load()
    return float(data.get(_key(mac), {}).get(str(int(ch)), {}).get("remaining_ml", 0.0))

def set_capacity(mac: str, ch: int, ml: float) -> None:
    if ml < 0:
        raise ValueError("capacity must be >= 0 ml")
    data = _load()
    chd = _ensure_channel(data, mac, ch)
    chd["capacity_ml"] = float(ml)
    # If remaining wasn't set, initialize to capacity
    chd.setdefault("remaining_ml", float(ml))
    _save(data)

def set_remaining(mac: str, ch: int, ml: float) -> None:
    if ml < 0:
        ml = 0.0
    data = _load()
    chd = _ensure_channel(data, mac, ch)
    chd["remaining_ml"] = float(ml)
    _save(data)

def dec_remaining(mac: str, ch: int, ml: float) -> None:
    """Decrease remaining by ml (clamped at 0)."""
    current = get_remaining(mac, ch)
    set_remaining(mac, ch, max(0.0, current - float(ml)))

If you want a couple of CLI hooks to manage this (set capacity/remaining, show values), I can drop those into chihirosctl.py too.

Martin11180, 2025-09-25:
Change to code
Entry possible: 25.6 or 0.2

Higher ml no longer need to be sent individually, higher than 25.6
I only changed it in set_dosing_pump_manuell_ml for now, you'll probably make changes later.

base_device.py

```
import array

from ...chihiros_doser_control import dosingpump
ML_CHECK = [25.6, 51.2, 76.8, 102.4, 128.0, 153.6, 179.2, 204.8, 230.4, 256.0, 281.6, 307.2, 332.8, 358.4, 384.0, 409.6, 435.2, 460.8, 486.4, 512.0, 537.6, 563.2, 588.8, 614.4, 640.0, 665.6, 691.2>

async def set_dosing_pump_manuell_ml(
    self,
    ch_id: Annotated[int, typer.Option(max=3, min=0)] = 0,
    # 0.1 ml units (1 == 0.1 ml). Allow up to 9999 == 999.9 ml.
    ch_ml: Annotated[float, typer.Option(min=0.2, max=999.9)] = None,
    ) -> None:
      if ch_ml <= 25.5:
         bucket_one = 0
         bucket_two = int( str(ch_ml).replace('.', ''))
      else :
         bucket_one = int(ch_ml / 25.6)
         if ch_ml in ML_CHECK:
            bucket_two = 0
         else:
            bucket_two = int( str( round( ch_ml - bucket_one * 25.6 ,2)).replace('.', ''))

        #print(bucket_one)
        #print(bucket_two)
        cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
            self.get_next_msg_id(),
            ch_id,
            bucket_one,
            bucket_two
        )
        #print(cmd)
        #x = []
        #for i in range(1, 39):
        #   a= round(25.6 * i,2)
        #   x.append(a)
        #print(x)

        #remaining = ch_ml
        #while remaining > 0:
        #    bucket = min(remaining, 256)  # ≤ 25.0 ml (250 units)
        #    print(bucket)
        #    cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
        #        self.get_next_msg_id(),
        #        ch_id,
        #        bucket,
        #    )
        await self._send_command(cmd, 3)
        #    remaining -= bucket
		
```

chihirosctl.py

```
from ..chihiros_doser_control import dosingpump

@app.command()
def set_dosing_pump_manuell_ml(
   device_address: str,
   ch_id: Annotated[int, typer.Option(max=3, min=0)] = 0,
   # 0.1 ml units (1 == 0.1 ml). Allow large totals; device code splits into buckets.
   ch_ml: Annotated[float, typer.Option(min=0.2, max=999.9)] = None,
 ) -> None:
     _run_device_func(
         device_address,
         ch_id=ch_id,
         ch_ml=ch_ml,
         func_name="set_dosing_pump_manuell_ml",
     )
```

dosingpump.py

```
def create_add_dosing_pump_command_manuell_ml(
    msg_id: tuple[int, int],
    ch_id: int,
    ch_ml_one: int,  # 0.1 ml units
    ch_ml_two: int,  # 0.1 ml units
) -> bytearray:
    if ch_ml_two < 0:
        raise ValueError("ch_ml_two must be >= 0 (0.1 ml units)")
    # [channel, ?, ?, ?, amount_units]
    return _create_command_encoding_dosing_pump(165, 27, msg_id, [ch_id, 0, 0, ch_ml_one, ch_ml_two])
```

Martin11180, 2025-09-25:
Good morning

In the parameters

165 27 [2, 127, 1, 0, 1, 104], the last two parameters are in ml.
I've tested a few things and it works.
I'm just not sure if everything works with float init str as well.

This way it can be sent in one go.

I haven't written any code for the 24-dose meter yet.

The typer issue probably requires a separator.

Martin11180, 2025-09-26:
I'm currently trying to sort out the code a bit. I think that's in your best interest.

I now have the CTL for the light and dosing pump separated.

The import section in the chihirosctl.py file
looks like this.

 ``` 

import asyncio
import inspect
from datetime import datetime
from typing import Any

import typer
from bleak import BleakScanner
from rich import print
from rich.table import Table
from typing_extensions import Annotated

from . import commands
from .device import get_device_from_address, get_model_class_from_name
from .weekday_encoding import WeekdaySelect

from ..chihiros_doser_control import dosingpump,chihirosdosesctl
from . import storage_containers as sc
from typing import List, Tuple

app = typer.Typer()
app.add_typer(chihirosdosesctl.app)

 ```

These files and folder are now in the chihiros_doser_control folder

chihiros_doser_control/chihirosdosesctl.py

```
import asyncio
import inspect
from datetime import datetime
from typing import Any

import typer
from bleak import BleakScanner
from rich import print
from rich.table import Table
from typing_extensions import Annotated
from typing import List, Tuple
from prettytable import PrettyTable, SINGLE_BORDER
from ..chihiros_led_control import commands
from ..chihiros_led_control import chihirosctl
from ..chihiros_led_control.device import get_device_from_address, get_model_class_from_name
from ..chihiros_led_control.weekday_encoding import WeekdaySelect

app = typer.Typer()

def _parse_params_tokens(tokens: List[str]) -> List[int]:
    out: List[int] = []
    for t in tokens:
        s = t.strip().lower()
        if s.startswith("0x"):
            v = int(s, 16)
        else:
            is_hex_byte = len(s) <= 2 and all(c in "0123456789abcdef" for c in s)
            v = int(s, 16) if is_hex_byte else int(s, 10)
        if not (0 <= v <= 255):
            raise typer.BadParameter(f"Parameter-Byte außerhalb 0..255: {t}")
        out.append(v)
    return out
[... 229 lines cut ...]
def create_switch_to_auto_mode_dosing_pump_command(
   msg_id: tuple[int, int],
   channel_id: int,
) -> bytearray:
  """Switch dosing pump to auto mode for a specific channel: payload = [channel_id, 0, 1]."""
  return _create_command_encoding_dosing_pump(165, 32, msg_id, [channel_id, 0, 1])

```
chihiros_doser_control
'__init__ copy.py'   __init__.py   protocol.py   __pycache__

In the device folder, I wanted to separate the classes from Light and Pump.
I'm stuck.

Folder Device 

base_device.py split dosing function

to doses_device.py
```
???????????????

```

I also have more thoughts and ideas, I hope you will add them as well, otherwise you will lose track of things.

MankiniChykan, 2025-09-27:
Sorry mate I have been a bit busy. I am opening up my repository on code spaces now and looking at these.

Martin11180, 2025-09-27:
No problem, I'm learning from it.
If it worked that way, it wouldn't be bad.
 ``` 
chihiros_rgb_ch4_control
             --- template.py
                      -- data (json intern)
             --- chihirochrgbch4ctl.py
             - device
                     --- rgbch4_device.py
chihiros_led_control
            --- template.py
                  -- data (json intern) 
             - device
                    - base_device.py  
 ```

MankiniChykan, 2025-09-27:
I already have the __init__.py and protocol.py separated.

Martin11180, 2025-09-27:
chihirochrgbch4ctl.py

 ``` 
@app.command()
def set_rgb_brightness_ch_4(
    device_address: str, brightness: Annotated[tuple[int, int, int, int], typer.Argument()]
) -> None:
    """Set brightness of a RGB light. mit 4 channel"""
    print(brightness)
    _run_device_func(device_address, brightness=brightness)

```   

template.py

```
@app.command()
def load_template_standart(
    device_address: str,
    model_art: str,
    template_name: Annotated[str, typer.Option()],
) -> None:
       red, green, blue, white = sc.template_standart_get(model_art, template_name)
       brightness = (red, green, blue, white)
       print(brightness)
       _run_device_func(device_address, brightness=brightness)

@app.command()
def set_template_standart(
    model_art: str,
    template_name: Annotated[str, typer.Option()],
    red: Annotated[int, typer.Option(min=0, max=140)],
    green: Annotated[int, typer.Option(min=0, max=140)],
    blue: Annotated[int, typer.Option(min=0, max=140)],
    white: Annotated[int, typer.Option(min=0, max=140)]):
    sc.set_template_standart(model_art, template_name, red, green, blue, white)
    typer.echo(f"Standart Template set: Model art={model_art} Template name={template_name} Red={red} Green={green} Blue={blue} White={white}")

@app.command()
def get_template_standart(
    model_art: str,
    template_name: Annotated[str, typer.Option()]):
    red, green, blue, white = sc.template_standart_get(model_art, template_name)
    typer.echo(f"Standart Template : Model art={model_art} Template name={template_name} Red={red} Green={green} Blue={blue} White={white}")

 ```

MankiniChykan, 2025-09-27:
Are you working from the repository or locally on your computer. Also are you now editing in the Home Assistant Custom Components?

Martin11180, 2025-09-27:
On Raspberry, I'm not yet using the Home Assistant Custom Components repository.

only CTL

MankiniChykan, 2025-09-27:
In my last commit I have split them. 

[b5f3222](https://github.com/MankiniChykan/chihiros-led-control/commit/b5f3222cf06195e5b181bf6c4e238b20fd1026c2)
.

That commit message is:

seperate ctl for doser

Key changes in that commit:

New module split
You introduced chihiros_doser_control/ with its own __init__.py and dosing.py.
That means light control (chihirosctl.py) and dosing control now live in separate namespaces.

Imports reorganized
chihirosctl.py no longer mixes dosing and light; it imports only device/light logic.
The doser code is imported cleanly from the new package.

CLI separation
Typer commands for dosing are now defined under chihiros_doser_control.dosing, not inside the main CLI for lights.
This prevents collisions and keeps argument parsing clearer.

Device handling
You added get_device_from_address and get_model_class_from_name helpers into the light side, while dosing has its own logic for parsing and sending.

In short: this commit is a structural refactor — it separates out dosing pump control from lighting control, giving each its own control entrypoint. This makes the project cleaner and prepares it for further dosing-specific development (like your float-init ml parameters you mentioned earlier).

MankiniChykan, 2025-09-27:
I only have the doser sorry mate

MankiniChykan, 2025-09-27:
> fehlt def _run_device_func(device_address: str, func_name: str | None = None, func=None, **kwargs):
> 
> not woring chihirosctl raw-dosing-pump XXXXX --cmd-id 165 --mode 27 --params 0 0 0 0 10 Got unexpected extra arguments (0 0 0 10)
> 
> working chihirosctl set-dosing-pump-manuell-ml XXXXX --ch-id 0 --ch-ml 10
> 
> working chihirosctl add-setting-dosing-pump XXXX 22:38 --ch-id 0 --weekdays wednesday --ch-ml 80
> 
> ```
> chihirosctl bytes-encode 5b010a00012200710042006300961b
> ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
> │                                                                                     Encode Message                                                                                    │
> ├─────────────────────────────────────────────────────────┬────────────┬─────────┬────────────────┬─────────────────┬────────────────┬──────┬────────────────────────────────┬──────────┤
> │                      Command Print                      │ Command ID │ Version │ Command Length │ Message ID High │ Message ID Low │ Mode │           Parameters           │ Checksum │
> ├─────────────────────────────────────────────────────────┼────────────┼─────────┼────────────────┼─────────────────┼────────────────┼──────┼────────────────────────────────┼──────────┤
> │ [91, 1, 10, 0, 1, 34, 0, 113, 0, 66, 0, 99, 0, 150, 27] │     91     │    1    │       10       │        0        │       1        │  34  │ [0, 113, 0, 66, 0, 99, 0, 150] │    27    │
> └─────────────────────────────────────────────────────────┴────────────┴─────────┴────────────────┴─────────────────┴────────────────┴──────┴────────────────────────────────┴──────────┘
> falsch :Decoded daily totals (ml): ch0=2892.8, ch1=1689.6, ch2=2534.4, ch3=3840.0
> richtig :Decoded daily totals (ml): ch0=11.30, ch1=6.60, ch2=9.90, ch3=15.00
> ```
> 
> import geändert dosingpump in folder chihiros_doser_control
> 
> from ..chihiros_doser_control import dosingpump
> 
> I think typer has problems with multi values ​​with space

I adjusted my files as follows:

### __init__.py
```
from __future__ import annotations

import voluptuous as vol
from homeassistant.core import HomeAssistant, ServiceCall
from homeassistant.helpers import device_registry as dr
from homeassistant.exceptions import HomeAssistantError
from bleak import BleakClient

from ..const import DOMAIN  # integration domain
from . import protocol as dp  # provides dose_ml(client, channel, ml)

# Accepts "2" or 2, "10.0" or 10.0 via Coerce; ranges match device limits.
DOSE_SCHEMA = vol.Schema({
    vol.Exclusive("device_id", "target"): str,
    vol.Exclusive("address", "target"): str,
    vol.Required("channel"): vol.All(vol.Coerce(int), vol.Range(min=1, max=4)),
    vol.Required("ml"):      vol.All(vol.Coerce(float), vol.Range(min=0.2, max=999.9)),
})

async def _resolve_address_from_device_id(hass: HomeAssistant, did: str) -> str | None:
    reg = dr.async_get(hass)
    dev = reg.async_get(did)
    if not dev:
        return None

    # 1) Native BT connection stored by HA
    for (conn_type, conn_val) in dev.connections:
        if conn_type == dr.CONNECTION_BLUETOOTH:
            return conn_val

    # 2) Identifiers used by this integration
    for domain, ident in dev.identifiers:
        if domain != DOMAIN:
            continue
        # a) explicit "ble:AA:BB:..."
        if isinstance(ident, str) and ident.startswith("ble:"):
            return ident.split(":", 1)[1]
        # b) (DOMAIN, <config_entry_id>) → hass.data[DOMAIN][entry_id].coordinator.address
        data_by_entry = hass.data.get(DOMAIN, {})
[... 123 lines cut ...]

async def dose_ml(client, channel_1based: int, ml: float) -> None:
    """
    Immediate, one-shot dose on the selected channel.

    Protocol encoding (confirmed by sniffing):
      params = [channel(0..3), 0x00, 0x00, ml_hi, ml_lo]
      where:
        ml_hi = floor(ml / 25.6)
        ml_lo = round((ml - ml_hi*25.6) * 10)    # 0.1 mL remainder

    Examples:
      11.3  -> (0,113)
      25.6  -> (1,0)
      51.2  -> (2,0)
      29.0  -> (1,34)
    """
    ch = max(1, min(int(channel_1based), 4)) - 1  # 0-based on wire
    ml = round(max(0.2, min(float(ml), 999.9)), 1)

    ml_hi, ml_lo = _split_ml_25_6(ml)
    pkt = _encode(CMD_MANUAL_DOSE, MODE_MANUAL_DOSE, [ch, 0x00, 0x00, ml_hi, ml_lo])
    await client.write_gatt_char(UART_RX, pkt, response=True)

```

MankiniChykan, 2025-09-27:
@Martin11180 Here’s the **fork-vs-upstream diff summary** (My fork at `main` around commit `b5f3222` vs `TheMicDiet:main`), grouped by status:

I have split the doser away from the light. I will now try add your other bits of code into mine.

### New files (6)

* `custom_components/chihiros/chihiros_doser_control/__init__.py` ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.patch)][1])
* `custom_components/chihiros/chihiros_doser_control/protocol.py` ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.patch)][1])
* `custom_components/chihiros/button.py` ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/number.py` ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/services.yaml` ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/chihiros_doser_control/__init__ copy.py` (looks like a stray copy; consider deleting) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])

### Modified files (7)

* `custom_components/chihiros/__init__.py` (platform selection, doser detection, service registration, options listener) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/config_flow.py` (adds OptionsFlow for enabled channels) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/light.py` (skip creating lights for doser devices) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/switch.py` (skip creating switches for doser devices) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `custom_components/chihiros/manifest.json` (adds DYDOSE patterns; deps tweaked; version bump) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `pyproject.toml` (Python version constraint) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])
* `setup.cfg` (python_requires + extras) ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain.diff)][2])

GitHub’s cross-fork compare shows **“Files changed 13”** for this diff, matching the 6 new + 7 modified above. ([[GitHub](https://github.com/TheMicDiet/chihiros-led-control/compare/main...MankiniChykan%3Amain)][3])

> I'm currently trying to sort out the code a bit. I think that's in your best interest.
> 
> I now have the CTL for the light and dosing pump separated.
> 
> The import section in the chihirosctl.py file looks like this.
> 
> ```
> 
> import asyncio
> import inspect
> from datetime import datetime
> from typing import Any
> 
> import typer
> from bleak import BleakScanner
> from rich import print
> from rich.table import Table
> from typing_extensions import Annotated
> 
> from . import commands
> from .device import get_device_from_address, get_model_class_from_name
> from .weekday_encoding import WeekdaySelect
> 
> from ..chihiros_doser_control import dosingpump,chihirosdosesctl
> from . import storage_containers as sc
> from typing import List, Tuple
> 
> app = typer.Typer()
> app.add_typer(chihirosdosesctl.app)
> ```
> 
> These files and folder are now in the chihiros_doser_control folder
> 
> chihiros_doser_control/chihirosdosesctl.py
> 
> ```
> import asyncio
> import inspect
> from datetime import datetime
> from typing import Any
> 
> import typer
> from bleak import BleakScanner
> from rich import print
> from rich.table import Table
[... 469 lines cut ...]
>     ) -> None:
>         query = dosingpump.create_query_container_status_command(self.get_next_msg_id(), ch_id)
>         await self._send_command(query, 3)
>         await asyncio.sleep(timeout_s)
>         payload = getattr(self, "_last_response", None)
>         frames = getattr(self, "_last_frames", None)
>         if payload is None and frames:
>             payload = b"".join(frames)
>         if payload is None:
>             typer.echo("Keine Antwort erfasst (Notify-Callback muss _last_response/_last_frames setzen).")
>             return
>         hexstr = " ".join(f"{b:02X}" for b in payload)
>         typer.echo(f"Raw response ({len(payload)} bytes): {hexstr}")
>         parsed = dosingpump.try_parse_container_status(payload)
>         if parsed:
>             typer.echo("Parsed container status:")
>             for e in parsed:
>                 typer.echo(
>                     f"  ch={e['channel']} total={e['total_ml']:.1f} ml (units={e['total_units']}) remaining={e['remaining_ml']:.1f} ml (units={e['remaining_units']}) at{e['offset']}"
>                 )
>         else:
>             typer.echo("Konnte Container-Status nicht parsen — Parser ggf. anpassen.")
> ```
> 
> I also have more thoughts and ideas, I hope you will add them as well, otherwise you will lose track of things.

Martin11180, 2025-09-27:
I don't know how to open a private thread on Github
where only you and I can write

TheMicDiet, 2025-09-27:
It's great that you're making progress with the dosing pump. When you have everything up and running, feel free to submit a PR.

Martin11180, 2025-09-27:
1

chihirosctl bytes-encode and raw_dosing_pump

They should ideally be located centrally; they have nothing to do with a device
and are actually helpful.

2 

open: “sniffer 24-hour dosage” (likely another mode — we’ll tag it when we see the frame)

24-hour doses

Minutes only: 00:59
Minimum dose: 0.2
Maximum dose: 999.9

Current setting

Daily dose: 36 ml
Time: 16 minutes
Day of the week: Any

````
90   4  [1]
90   9  [25, 9, 4, 8, 23, 20]
90   9  [25, 9, 4, 8, 23, 20]
165  4  [4]
165  4  [5] 
165  32 [2, 0, 1]								
165	 27 [2, 127, 1, 0, 1, 104] 					   
165  21 [2, 1, 0, 16, 0, 0] 		

165 32 [channel-id, Make up for missed dose, Active = 1 or Inactive = 0]
165 27 [channel-id, weekday, ?, timer completed for today 0 = No 1 = fertig, ml, ml]
165 21 [channel-id, timer-type, hour, minutes, ?, ?]
````

the code still needs to be written

3
 open: “sniffer custom dosage” (probably a schedule bank read)

I'll try to sniff around this weekend

4
 open: “sniffer timer dosage” (readback of scheduled time(s) and ML per entry)
I'll try to sniff around this weekend

5 
chihirosctl raw-dosing-pump XXXXX --cmd-id 165 --mode 27 --params 0 0 0 0 10 
fix --params

6
Send entire container to json HA and CLI

7
daily consumption HA and CLI query

Then we should have everything we need regarding the pump, and you're making the final progress.

I'll try to sniff out the rest this weekend, but I can't hear the pump anymore.

MankiniChykan, 2025-09-27:
I created a new service to set auto dose it is not working properly yet.
Need some more frames around this one.

[image]

Also Added read daily totals which fails.

[image]

@Martin11180 have a good look at the files in here. I have not wired up your doser_device.py and dosingpump.py files fully yet.

With the files you posted do you have those functions working?

Martin11180, 2025-09-27:
I don't quite understand your question about the files.
If you build the function using CTL, I can help you see where the error might be.
HA isn't mine yet.

Martin11180, 2025-09-27:
I can not test the commands via ctl because the import wiring is still missing in the ctl main file

Martin11180, 2025-09-27:
I  can't get any further, I guess I have to dismantle it

MankiniChykan, 2025-09-27:
Did you get this running in Home Assistant?
You need to add this as a custom repository[https://github.com/MankiniChykan/chihiros-led-control]( https://github.com/MankiniChykan/chihiros-led-control)

MankiniChykan, 2025-09-28:
@Martin11180 can you write a list of exactly what you do have working and the actual method you are using to get them working?
- chihiros_doser_control/chihirosdosesctl.py
- chihiros_doser_control/dosingpump.py
- doses_device.py

Do your files work together?

I only have the manual dosing working.

Martin11180, 2025-09-28:
chihirosctl set-dosing-pump-manuell-ml xxxx --ch-id 0 --ch-ml 10 workin

base_device.py
```
from ...chihiros_doser_control.protocol import _split_ml_25_6

    async def set_dosing_pump_manuell_ml(
        self,
        ch_id: Annotated[int, typer.Option(max=3, min=0)] = 0,
        ch_ml: Annotated[float, typer.Option(min=0.2, max=999.9)] = 0.2,
    ) -> None:
        """Immediate dose on channel using 25.6-bucket + 0.1 remainder (single frame)."""
        hi, lo = _split_ml_25_6(ch_ml)          # (hi = floor(ml/25.6), lo = 0.1-ml remainder)
        cmd = dosingpump.create_add_dosing_pump_command_manuell_ml(
            self.get_next_msg_id(), ch_id, hi, lo
        )
        await self._send_command(cmd, 3)
```

If I insert it again as before, the function works again so the class call of DoserDevice is missing somewhere

MankiniChykan, 2025-09-28:
That is only manual dosing though isn't it?

Martin11180, 2025-09-28:
no 
 chihirosctl add-setting-dosing-pump  --ch-id 0 --weekdays monday --ch-ml 20
Fallback doesn't support add_setting_dosing_pump

Martin11180, 2025-09-28:
> **Handbuch One Shot Dosing** - JA Wir haben diese Arbeit.
> **Senden einer automatischen Dosis (Zeitplan)**- NEIN Ich habe diese Arbeit nicht.

But if I write the function, you'll implement it, right?
That one wasn't directed at you.

Martin11180, 2025-09-28:
following params come as notification
[1, 255, 255, 255, 200, 9, 118] 	
[18, 153, 1, 174, 7, 51, 0, 148] 
[1, 47, 0, 0, 0, 27, 0, 41] 
[1, 255, 255, 255, 200, 9, 119] 
[91, 6, 48, 0, 1, 254, 7, 12, 10, 0, 5, 0, 0, 0, 0, 0, 1, 47, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6, 0, 0, 0, 0, 0, 0, 27, 0, 7, 0, 0, 0, 0, 0, 0, 41, 30, 35, 2, 4, 0]

Martin11180, 2025-09-28:
Got it working.
Need to make the following changes.
Unfortunately, there are still some errors in our code. See below.

```
custom_components/chihiros/chihiros_doser_control/doser_device.py

from typing_extensions import Annotated
-from ..chihiros_led_control.device.base_device import BaseDevice 
+from abc import ABC, ABCMeta

-class DoserDevice(BaseDevice):
+class DoserDevice(ABC):
```

```
custom_components/chihiros/chihiros_led_control/device/base_device.py

from ..weekday_encoding import WeekdaySelect, encode_selected_weekdays
+from ...chihiros_doser_control.doser_device import DoserDevice

class BaseDevice(ABc):
+class BaseDevice(DoserDevice):
```

chihirosctl set-dosing-pump-manuell-ml DF:2A:88:E7:8A:90 --ch-id 0 --ch-ml 10.5

chihirosctl set-dosing-pump-manuell-ml
Missing argument 'DEVICE_ADDRESS'.   

chihirosctl set-dosing-pump-manuell-ml DF:2A:88:E7:8A:90   
not return Missing argument 'CH-ID'.

chihirosctl set-dosing-pump-manuell-ml DF:2A:88:E7:8A:90 --ch-id 0 
not return Missing argument 'CH-ML'.

chihirosctl add-setting-dosing-pump DF:2A:88:E7:8A:90
not return Missing argument 'DEVICE_ADDRESS'. 

chihirosctl add-setting-dosing-pump DF:2A:88:E7:8A:90 08:15
Missing argument 'PERFORMANCE_TIME:[%H:%M]'. 

chihirosctl add-setting-dosing-pump DF:2A:88:E7:8A:90 08:15
not return Missing argument 'CH-ID'

chihirosctl add-setting-dosing-pump DF:2A:88:E7:8A:90 08:15 --ch-id 0
not return Missing argument 'WEEKDAYS'

chihirosctl add-setting-dosing-pump DF:2A:88:E7:8A:90 08:15 --ch-id 0 --weekdays everyday
not return Missing argument 'CH-ML' 

chihirosctl raw-dosing-pump 
Missing argument 'DEVICE_ADDRESS'.  

chihirosctl raw-dosing-pump DF:2A:88:E7:8A:90
not return Missing argument 'CH-ID'.

chihirosctl raw-dosing-pump DF:2A:88:E7:8A:90 --cmd-id 165
not return Missing argument 'MODE'.

chihirosctl raw-dosing-pump DF:2A:88:E7:8A:90 --cmd-id 165 --mode 27
not return Missing argument 'PARAMS'.
Error : TypeError: BaseDevice.raw_dosing_pump() got an unexpected keyword argument 'p'

chihirosctl raw-dosing-pump DF:2A:88:E7:8A:90 --cmd-id 165 --mode 27 0 0 14 02 0 0
Error:│ Got unexpected extra arguments (0 0 14 02 0 0)

MankiniChykan, 2025-09-29:
Just made this change to the repository. Re-Download.

Yep — your CLI isn’t seeing the doser commands because they live in a separate Typer app. Mount the doser app under a `doser` subcommand.

Replace your `custom_components/chihiros/chihiros_led_control/chihirosctl.py` with this (only 3 small additions marked “NEW”):

```python
"""Chihiros led control CLI entrypoint."""

import asyncio
import inspect
from datetime import datetime
from typing import Any

import typer
from bleak import BleakScanner
from rich import print
from rich.table import Table
from typing_extensions import Annotated

from . import commands
from .device import get_device_from_address, get_model_class_from_name
from .weekday_encoding import WeekdaySelect

# NEW: import the doser Typer app
from ..chihiros_doser_control.doser_device import app as doser_app  # NEW

app = typer.Typer()

msg_id = commands.next_message_id()

def _run_device_func(device_address: str, **kwargs: Any) -> None:
    command_name = inspect.stack()[1][3]

    async def _async_func() -> None:
        dev = await get_device_from_address(device_address)
        if hasattr(dev, command_name):
            await getattr(dev, command_name)(**kwargs)
        else:
            print(f"{dev.__class__.__name__} doesn't support {command_name}")
            raise typer.Abort()

    asyncio.run(_async_func())

@app.command()
def list_devices(timeout: Annotated[int, typer.Option()] = 5) -> None:
    """List all bluetooth devices.

    TODO: add an option to show only Chihiros devices
    """
    table = Table("Name", "Address", "Model")
    discovered_devices = asyncio.run(BleakScanner.discover(timeout=timeout))
    for device in discovered_devices:
        model_name = "???"
        if device.name is not None:
            model_class = get_model_class_from_name(device.name)
            if model_class.model_code:  # type: ignore
                model_name = model_class.model_name  # type: ignore
        table.add_row(device.name, device.address, model_name)
    print("Discovered the following devices:")
    print(table)

@app.command()
def turn_on(device_address: str) -> None:
    """Turn on a light."""
    _run_device_func(device_address)

@app.command()
def turn_off(device_address: str) -> None:
    """Turn off a light."""
[... 89 lines cut ...]

# NEW: mount the doser commands at `chihirosctl doser ...`
app.add_typer(doser_app, name="doser")  # NEW

if __name__ == "__main__":
    try:
        app()
    except asyncio.CancelledError:
        pass
```

### How to use

* Keep `click==8.1.8` (or upgrade Typer too).
* Reinstall the package: `pip install -e .`
* Run:

  ```bash
  chihirosctl doser --help
  chihirosctl doser set-dosing-pump-manuell-ml AA:BB:CC:DD:EE:FF --ch-id 0 --ch-ml 10.5
  chihirosctl doser add-setting-dosing-pump AA:BB:... 08:15 --ch-id 0 --weekdays everyday --ch-ml 36
  chihirosctl doser raw-dosing-pump AA:BB:... --cmd-id 165 --mode 27 0 0 14 2 0 0
  ```

That’s it—your doser commands are now part of the same CLI.

Martin11180, 2025-09-29:
I've now deleted everything and recreated it, but unfortunately I still get the message
MixedAnnotatedAndDefaultStyleError: Cannot specify `Argument` in `Annotated` and default value together for 'params'

MankiniChykan, 2025-09-29:
> I've now deleted everything and recreated it, but unfortunately I still get the message MixedAnnotatedAndDefaultStyleError: Cannot specify `Argument` in `Annotated` and default value together for 'params'

Try again now I am just working through this chihirosctl 

[image]

Got rid of the MixedAnnotatedAndDefaultStyleError: Cannot specify Argument in Annotated and default value together for 'params'

[image]

MankiniChykan, 2025-09-29:
I am slowly getting there.

Martin11180, 2025-09-29:
No, manual dosing is not possible.
If it runs automatically, yes.
Setting the timer is not possible either.
The same is true for the LED.
I don't know what's in the bytes.

[1, 255, 255, 200, 9, 118]
[18, 153, 1, 174, 7, 51, 0, 148]
[1, 47, 0, 0, 0, 27, 0, 41]
[1, 255, 255, 255, 200, 9, 119]
[91, 6, 6, 48, 0, 1, 254, 7, 12, 10, 0, 0, 0, 0, 0, 0, 0, 1, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 30, 2, 4, 0]

Martin11180, 2025-09-29:
I'm currently trying to figure out if it's possible to search in two places in this call:
for _name, _obj in inspect.getmembers(sys.modules[__name__]):  and  chihiros_ch4_control/device
that the classes of ch3 and ch4 are separated

Martin11180, 2025-09-30:
I haven't tested it yet because I wanted to implement it first so I could delete my changes and have them updated with mine.

It only works with one model for now. Something needs to be changed in the class so that all models go from this folder.

With
chihirosctl set-rgb-brightness-ch-4 xxx 30 30 30 30 I can control the RGB, but the total cannot exceed 400% because of a check before sending, so 30 + 30 + 30 + 30
The maximum per channel is 140%
limited.

I just don't know if I've sent all the changes here yet. I'll have to deal with the pull-request issue from
VS Studio.

What doesn't work in HA is that there's still a lot of adjustment to be made in the int woll, but one at a time.

[3 more replies]

closed 2026-09-13 by TheMicDiet