TheMicDiet/chihiros-led-control #69: Support for Commander X

pacorreia, 2025-08-31:
Hi,

Any chance of adding support for Commander X?

Device ID: DYCOMD47C40B8B811

The plugin detected it, but only red and white toggle seem to work,in the wrong, but they work byt turning them on/off.

This device support color changing and brightness control.

I'll more than happy to help providing any help

Thanks

TheMicDiet, 2025-09-27:
Hi @pacorreia, I added the possibility to select the device type for commander 1 and 4

closed 2025-09-27 by TheMicDiet

pacorreia, 2025-09-27:
> Hi [@pacorreia](https://github.com/pacorreia), I added the possibility to select the device type for commander 1 and 4

awesome, going to test it asap

pacorreia, 2025-09-28:
> > Hi [@pacorreia](https://github.com/pacorreia), I added the possibility to select the device type for commander 1 and 4
> 
> awesome, going to test it asap

@TheMicDiet I tested by choosing a white led,and it doesn't work, neither brightness control shows up

would help to capture packets or else?

pacorreia, 2025-09-29:
> I don't get it. If you choose white there will be one toggle for white. Are you saying there is no toggle at all?

Yep there's a toggle for white, but it doesn't work.
Tried the other profiles they don't work either.

I'll reset the commander unit and try again